cta-observatory/ctapipe #875: Define camera coordinate systems used by ctapipe

mackaiver, 2018-12-11:
And add them to the documentation. One should be used internally. The others should have names by which they can be identified.

thomasgas, 2018-12-14:
These are the geometries used now in ctapipe. I plotted both the x and y coordinates as taken from the ctapipe and the "flipped coordinates": for cameras on DC optics x goes to -y and y goes to -x, whereas for SC telescopes x goes to -y and y goes to x (is this true? @kosack can you confirm this).
@GernotMaier can you also have a look at these plots?
Here you are some images:
[image]
[image]
[image]
[image]
[image]
[image]

GernotMaier, 2018-12-16:
This is always difficult!

Please have a look at the MC parameter descriptions available for each telescope type in:

https://forge.in2p3.fr/projects/cta_analysis-and-simulations/repository/show/Simulations/MCModelDescription/trunk/reports

There is a figure in each report (e.g. Fig 6 in the LST report) with the coordinate systems explicitly in the figure. Am adding @orelgueta to this issue - he produced these figures and thought quite a while about the coordinate systems.

MaxNoe, 2018-12-17:
Maybe this also helps:

[image]

MaxNoe, 2018-12-17:
On the `fix_coordinate_usage` branch, do

```python
import matplotlib.pyplot as plt
from astropy.time import Time
from astropy.coordinates import SkyCoord, EarthLocation
import astropy.units as u
from ctapipe.coordinates import CameraFrame, HorizonFrame
from ctapipe.visualization import CameraDisplay
from ctapipe.instrument import CameraGeometry

location = EarthLocation.of_site('Roque de los Muchachos')
obstime = Time('2018-11-01T04:00')

crab = SkyCoord.from_name("crab nebula")

horizon_frame = HorizonFrame(location=location, obstime=obstime)
pointing = crab.transform_to(horizon_frame)
camera_frame = CameraFrame(
    telescope_pointing=pointing,
    focal_length=u.Quantity(28, u.m),
    obstime=obstime,
    location=location,
)

cam = CameraGeometry.from_name('LSTCam')
fig, ax = plt.subplots()
display = CameraDisplay(cam, ax=ax)

ax.set_title(
    f'La Palma, {obstime}, az={pointing.az.deg:.1f}°, zenith={pointing.zen.deg:.1f}°'
)

for i, name in enumerate(['crab nebula', 'o tau', 'zet tau']):
    star = SkyCoord.from_name(name)
    star_cam = star.transform_to(camera_frame)

    x = star_cam.x.to_value(u.m)
    y = star_cam.y.to_value(u.m)

    ax.plot(x, y, marker='*', color=f'C{i}')
    ax.annotate(
        text=name, xy=(x, y), xytext=(5, 5),
        textcoords='offset points', color=f'C{i}'
    )

plt.show()
```

MaxNoe, 2018-12-18:
I have this for fact (`pip install pyfact` to play along):

```python
from fact.coordinates import CameraFrame
from fact.plotting import camera, mark_pixel
from astropy.time import Time
from astropy.coordinates import SkyCoord, AltAz
import matplotlib.pyplot as plt
import astropy.units as u
import numpy as np

obstime = Time('2018-12-18T02:00')
alt_az = AltAz(obstime=obstime, location=CameraFrame().location)

crab = SkyCoord.from_name('crab nebula')
pointing = crab.transform_to(alt_az)
camera_frame = CameraFrame(obstime=obstime, pointing_direction=pointing)

fig, ax = plt.subplots()
camera(np.zeros(1440), ax=ax, picker=True)

mask = np.zeros(1440, dtype=bool)
mask[464] = True
mark_pixel(mask, ax=ax)

ax.set_title(
    f'La Palma, {obstime}, az={pointing.az.deg:.1f}°, zenith={pointing.zen.deg:.1f}°'
)

for i, name in enumerate(['crab nebula', 'zet tau', 'o tau']):
    star = SkyCoord.from_name(name)
    star_cam = star.transform_to(camera_frame)

    x = star_cam.x.to_value(u.mm)
    y = star_cam.y.to_value(u.mm)

    ax.plot(x, y, marker='*', color=f'C{i}')
    ax.annotate(
        text=name, xy=(x, y), xytext=(5, 5),
        textcoords='offset points', color=f'C{i}'
    )

plt.show()
```

Foto of the actual camera I took myself standing at the dish, looking onto the camera.
Notice the missing pixels in the top and bottom row on the left side?

[image]

And results of the script:

[image]

This brings me to the conclusion, I have to transform the FACT camera coords like this to get ctapipe camera coords:

`x -> -x`, `y -> -y`.

This contradicts what @moralejo told me (He said `x -> -y`, `y -> -x`

MaxNoe, 2018-12-18:
I came to the same conclusion for MAGIC.

I analysed a Crab run MAGIC took yesterday.

This is with the transformation MAGIC → ctapipe `x→-x, y→ -y`:

[image]

moralejo, 2018-12-19:
> I came to the same conclusion for MAGIC.
>
> I analysed a Crab run MAGIC took yesterday.
>
> This is with the transformation MAGIC → ctapipe x→-x, y→ -y:
I think this cannot be the transformation from HESS to MAGIC camera system,
because the HESS system has the "right orientation" of the xy axes (i.e.
you go from +x to +y via a 90 deg counter-clockwise rotation) when one
looks at the camera *from behind it *(i.e. from above when telescope points
at zenith).  The MAGIC system has the right orientation when looking from
the dish.

It seems to me that with x  => -x    y=> -y    you are just "applying the
reflection" on the mirror.

[image: magic_cam_max]
> <https://user-images.githubusercontent.com/5488440/50171518-f529a880-02f2-11e9-9d0e-a494cfbcc090.png>

-- 

Abelardo Moralejo Olaizola
Institut de Física d'Altes Energies
Tel : +34 931641662
Fax: +34 935811938
Avís - Aviso - Legal Notice - (LOPD) - http://legal.ifae.es

MaxNoe, 2018-12-19:
I came to the conclusion, that ctapipe implements the transformations wrong.

These are diffuse gammas from simtel. If I do the transform x -> y, y-> x, then the source positions look reasonable, which would also match the transformation @moralejo described

MaxNoe, 2018-12-19:
[image]
[image]
[image]
[image]
[image]
[image]

MaxNoe, 2018-12-19:
Ah and just to make it clear, I refer to the state in #896, not the current master.

MaxNoe, 2018-12-19:
I think I found the reason, the redifinition of telescope frame and nominal frame also use x and y as coordinates, but x is along the azimuth and y along altitude, which is not the case for hess camera coordinates.

I will rename the members of telescope frame and nominal frame to `delta_az` and `delta_alt` to make that clear.

thomasgas, 2019-01-30:
@MaxNoe I copy & pasted the code you put here (https://github.com/cta-observatory/ctapipe/issues/875#issuecomment-447797109).
I just had to solve a small bug in plotting
```
ax.annotate(
   s=name, xy=(x, y), xytext=(5, 5),
   textcoords='offset points', color=f'C{i}',
)
```
to 
```
ax.annotate(
    text=name, xy=(x, y), xytext=(5, 5),
    textcoords='offset points', color=f'C{i}'
)
```
but then the image has x and y flipped with respect to yours:
[image]
SInce you mentioned that a x <-> y substitution was giving a correct reconstruction (and it seems you have implemented this), so I'll just edit the image you put in this PR, right?

thomasgas, 2019-01-30:
wrong button and I closed the issue.

MaxNoe, 2019-11-25:
done in #1085

closed 2019-11-25 by MaxNoe