cta-observatory/ctapipe #881: Usage of Astropy units.

mackaiver, 2018-12-11:
Use `u.Quantity(5, u.m)` and `<some_quantity>.to_value(u.TeV)` as much as possible. It is more efficient than `5 * u.m` and `<some_quantity>.to(u.TeV).value` .

thomasgas, 2018-12-11:
Some benchmarks:

```
%timeit a = u.Quantity(5, u.m)
4.27 µs ± 35.4 ns per loop (mean ± std. dev. of 7 runs, 100000 loops each)
%timeit b = 5 * u.m
6.73 µs ± 117 ns per loop (mean ± std. dev. of 7 runs, 100000 loops each)
```
==> 50% faster!
```
%timeit a.to_value(u.cm)
4.8 µs ± 62 ns per loop (mean ± std. dev. of 7 runs, 100000 loops each)
%timeit b.to(u.cm).value
9.74 µs ± 94 ns per loop (mean ± std. dev. of 7 runs, 100000 loops each)
```
==> 2x faster!!!

kosack, 2019-03-20:
from Axel Donath's talk on p44 here: https://indico.cern.ch/event/783425/contributions/3325103/attachments/1815239/2966497/gammapy-pygamma-2019.pdf

Should change all places that coordinates are used to the following patterns, which are faster due to less copying of data:

instantiation:
```py3
q = Quantity(value, unit='cm-2 s-1')  # old way, slower
q = Quantity(value, unit='cm-2 s-1', copy=False) # faster no copy
q = value << 'cm-2 s-1' # same as previous, since Astropy 3.1
```

conversion:
```py3
value = q.to("cm-2 s-1") # old way, slower
value = q.to_value("cm-2 s-1") # new way with no copy
```

MaxNoe, 2019-03-20:
> q = value << 'cm-2 s-1' # same as previous, since Astropy 3.1

I really don't like this syntax. Bit shifting by a unit?

So maybe stick to the more verbose one.

vuillaut, 2019-03-20:
> > q = value << 'cm-2 s-1' # same as previous, since Astropy 3.1
> 
> I really don't like this syntax. Bit shifting by a unit?
> 
> So maybe stick to the more verbose one.

Agreed, `q = Quantity(value, unit='cm-2 s-1', copy=False)` is more clear.

kosack, 2019-03-25:
> I really don't like this syntax. Bit shifting by a unit?

no idea why they chose this, but I guess it was just the least-important operator available (you should never bit-shift anything that has a quantity attached anyhow).  Still it's a bit ugly, even if it saves some typing.

kosack, 2019-03-25:
actually, I just tried the shift method, and I think the presentation had it a bit wrong: you can't shift a value by a string representation unless the LHS or RHS is already a Quantity:
```py3
x = 12 * u.m/u.s # works, but I guess is slower
x << 'cm/hr'   # conversion works -> Quantity
12 << 'cm/hr' # doesn't work, neither side is Quantity
12 << u.m/u.s # creates a quantity 12 m/s
```

So I think it is quite confusing, and probably should just use the

adonath, 2019-04-03:
@kosack Yes, you're right. Assigning a unit string with the `<<` obviously doesn't work (how should it if there is no units class involved...). So the correct example would be `value << u.Unit("cm-2 s-1")`. I'll correct and re-upload the slides.

I also don't like the `<<` operator much. In Gammapy we've changed consistently to the `Quantity(value, "cm22 s-1", copy=False)` and `q.to_value("cm-2 s-1")` pattern.

maxnoe, 2020-10-26:
I think we fixed this in most of the places and are aware of it for PR reviews of new code. Closing.

closed 2020-10-26 by maxnoe