cta-observatory/ctapipe #886: Move variables in SubarrayDescription

mackaiver, 2018-12-12:
Move the variables in the property to be class members so we do not recalculate them every time.

https://github.com/cta-observatory/ctapipe/blob/9e3e85f61bd7c70e9665b30f17bd2d9e1aeebcb7/ctapipe/instrument/subarray.py#L101

We can just save these positions as a skycoord.

thomasgas, 2018-12-12:
Just moving the tel_coords function (which is now a property) in the __init__ is not easy-peasy since the coordinates and the ids are built just after the first event has been loaded by hessio.
The new pyeventio solves this problem by loading the telescopes informations from the header of the file and not after the first event is loaded. I'll do this after pyeventio has become default.

maxnoe, 2020-10-26:
This is solved now since the simtel  event source directly constructs the subarray.

closed 2020-10-26 by maxnoe