cta-observatory/ctapipe #888: Make CameraCalibrator choose calibration based on Camera ID, not file type

kosack, 2018-12-12:
Right now, CameraCalibrator constructs the calibrator based on the type of the file (as read by the EventSource).  In the future we will have the raw data of multiple cameras in the same format, so we should really choose the calibration type from the camera ID (like "NectarCam") which you can get from the SubarrayDescription that is loaded from an EventSource currently.

kosack, 2018-12-12:
TO be more clear, it should be something like this:

```py3
#This:
calib = CameraCalibrator(subarray_description=event.inst.subarray)

# rather than this:
calib = CameraCalibrator(event_source=source)
``

kosack, 2018-12-12:
Internally it may set up multiple calibrators, covering all telescope types (it could use `subarray.telescope_types` (which is something like `['SST-ASTRI:ASTRICam', 'LST:LSTCam', 'MST:FlashCam']` to decide which ones to use.  The SubarrayDescription also allows mapping of tel_id to type, via `ids = sub.get_tel_ids_for_type("SST-ASTRI:ASTRICam")`

dneise, 2019-01-24:
Misclick

MaxNoe, 2020-03-20:
I think this got fixed in #1228

closed 2020-03-20 by MaxNoe