cta-observatory/ctapipe #901: Missing test coverage for ImPACT

MaxNoe, 2018-12-14:
While refactoring the coordinates, I came across large parts of the ImPACT code that are not unit tested,
as I changed things ImPACT was using but no tests failed.

thomasgas, 2018-12-14:
if i remember correctly some tests needs the templates to be tested. but maybe @ParsonsRD can comment on this.

ParsonsRD, 2018-12-14:
Yes we need to implement a test which basically runs through the full reconstruction, this is difficult with out including a template set in the ctapipe package though...

thomasgas, 2018-12-14:
maybe we could just have one small template (or a few) produced at a certain energy and a small MC file to test it against (at the same energy). We could test the reconstruction just at different energies (100 GeV and 1 TeV) in order to limit the size of the template and the tests for ImPACT. Make sense?
A "monochromatic MC file" might be useful also in other cases.

MaxNoe, 2018-12-14:
The test files can be pretty crappy to just test if the code works. For physics performance we anyway need a lot of data.

ParsonsRD, 2018-12-14:
One think we could do is to just create a dummy template class that returns only zeros and use that. It would at least show the the likelihood calculation is working properly etc. I think performing a full fit is a bit beyond the scope of the tests.

kosack, 2019-01-08:
I think the idea of a very simplistic template (something small we can include in the test suite - with very wide binning, and a dummy model like a gaussian), or as @ParsonsRD said, a class the emulates one might be fine for most tests.

maxnoe, 2021-05-11:
We need a `Tool` that can actually create the template file and then we can test it's application as well. A fixed set of template files is not enough.