cta-observatory/ctapipe #905: sst-1m labelled as SC

thomasgas, 2018-12-14:
Using a simtel file from Prod3b (`gamma_20deg_0deg_run100___cta-prod3_desert-2150m-Paranal-merged.simtel.gz`) I just found out that all telescopes SST-1M are labelled as Schwarzschild-Couder (SC) instead of Davies-Cotton (DC).
With this code:
```
optics = events[2].inst.subarray.tel[430].optics
camera = events[2].inst.subarray.tel[430].camera

print(camera)
print(optics)
print(optics.mirror_type)
```
I get:
```
DigiCam
SST-1M
SC
```
And this is true for all the telescopes:
```
event_table = event.inst.subarray.to_table()
event_table[event_table['tel_subtype'] == "1M"]["mirror_type"]  == "SC"
```
is true for all the 158 SST-1M in the hyper-array in Prod3b.

thomasgas, 2018-12-18:
@bregeon and/or @GernotMaier can you comment on this?

MaxNoe, 2018-12-18:
This does not come from the mcs themselfs but from the guessing lookuptable in ctapipe:

https://github.com/cta-observatory/ctapipe/blob/16457f090dd50d4c2c2b2b0a12dfabeb2517c7f5/ctapipe/instrument/optics.py#L12

thomasgas, 2018-12-18:
ok thanks @MaxNoe! i'll do a PR to solve this issue.

thomasgas, 2018-12-18:
solved with #908.

closed 2018-12-18 by thomasgas