cta-observatory/ctapipe #907: Peakpos calculation

thomasgas, 2018-12-18:
the peakpos image that we get out now in ctapipe is not very useful for small images since the pixels that contains photons from the shower are very few and they all have the same time value (not very useful to compute a gradient).
@kosack you also explained to me some details (that now i don't remember quite well) but it would be nice if you write down here how it's done and how it could be implemented to have a more useful peakpos image to be used to compute the time gradient.

MaxNoe, 2018-12-18:
You could take the weighted average of the waveform, I think this is done in MAGIC.

So `np.average(np.arange(num_samples), weights=waveform)`

For the production case, you should not create the arange array in each iteration, but once.

kosack, 2019-01-08:
The problem right now is the current peakpos (and for that matter the trace integration) is ulta simplistic, and does no interpolation. The idea, I believe (@watsonjj would be able to comment more) was to keep it to simple sums and implement interpolation as a pre-processing step in a WaveformCleaner, effectively generating more time bins by oversampling/filtering. 

However, as we discussed at the meeting, that is somewhat of a waste of CPU time and memory, since it then makes the pixel data much larger, so we should consider implementing a more fancy methods for integration and time calculations that are floating-point (i.e. not tied to integer sample numbers).  The question is how easy it is to do without being too slow or too complex code.    The simplest would be to create a 1D interpolator for each pixel's waveform, and then do true numerical integration.  However, some knowledge of signal processing and the hardware is needed to choose the optimal interpolation (in the end it may actually be more accurate to apply the correct filter to oversampled data).  I'd guess that for GHz samplers, simple linear interpolation may be fine, but for the 250 MHz ones, more complex signal filtering may be better.

watsonjj, 2019-01-08:
Originally the idea was that waveform cleaning (which could include things like filtering and upsampling) was a separate step before the charge extraction. Therefore different combinations between WaveformCleaners and ChargeExtractors could be flexibly chosen. However, I have also come to the realisation that even this is a bit strict as some charge extraction approaches don't really conform to this split. Also conforming to this split may cause a waste in CPU time and memory, as you said. 

I personally have been considering this for a while, and I have some up with two potential improvements to the current charge extraction procedure:
1. further splitting of the steps (Cleaning, PeakFinding, Upsampling, Extraction)
2. perhaps we abandon allowing flexible combination of techniques at runtime, and instead have a list of hardcoded ChargeExtractors containing all of the steps. For example, a CCNeiChargeExtractor class might do:

   - a cross correlation on the waveform using the reference pulse shape,
   - then a neighbour-peak finding, 
   - and then extracts the single cross-correlated sample at the peak time

   Whereas a WindowLocalChargeExtractor would:

   - perform a local peak finding per pixel 
   - integrate within a window using the peak time (like the existing LocalPeakIntegrator)

   And a PolynomialFitChargeExtractor could:

   - Fit a nth degree polynomial to the waveform
   - Extract the peakpos and charge from the polynomial description.

   It is clear that these three valid charge extraction approaches do not conform to the same internal steps. 

I am inclined to switch to the second procedure for charge extraction - we don't necessarily need full flexibility at runtime, and could instead operate with many hardcoded charge extraction approaches. But I would be curious about other suggestions and opinions.

With regards to the peakpos, I believe @ParsonsRD had some opinions on how the current Charge Extractors could be changed so peakpos is calculated as a float and is therefore more useful. It may be a smaller improvement we can do before we completely revamp the Charge Extraction procedure.

kosack, 2019-01-08:
I think I like that approach - you can still structure the various algorithm "bricks" as separate functions, and just have the classes be some combination of them.  That would reduce the complexity of choosing each part of the algorithm to a set of "pre-packaged" choices, but still allow code-reuse.  That's probably sufficient for our needs, since I doubt we will change methods so often once a good one is chosen.

closed 2019-10-31 by watsonjj