cta-observatory/ctapipe #915: Issues with Factories

dneise, 2019-01-08:
I had the impression our Component Factories had some issues (mainly a gut feeling), so I took a closer look and I believe I identified a few real issues, which are more than just my gut telling me that our current implementation is too complex. However the main driver of this issue is a persistent feeling of complexity.

I understand our current implementation of `Factory` in `core/factory.py` is driven mainly by the following use case: "Users of `Tool`s should be able to choose and configure specific implementations of certain tasks by name". 

Another important use case is certainly letting the factory decide what is the "correct" implementation for that task at hand, this is e.g. the case when the correct event source is automatically selector from the url, or when the correct extractor is selected based on the given event source.

So I set out to find an alternative implementation, which fulfills these use cases while not using meta programming.

The other issues I found in the process were these:

In case multiple factory products define traitlets of the same name, then `FactoryMeta` silently keeps only one of them and silently drops the rest. I do not know what happens if traitlets of the same name but different type are defined. 

As a consequence of this users cannot inspect different defaults of factory products, since the factory exposes only the first.

Well I worked a bit on an alternative implementation and will open a PR now for discussion.
I do not assume that this first try will already be better than our current implementation, just different. I believe it can be healthy to see different versions in order to have a fruitful discussion.

watsonjj, 2019-01-08:
Related to #821

watsonjj, 2019-01-09:
> In case multiple factory products define traitlets of the same name, then FactoryMeta silently keeps only one of them and silently drops the rest. I do not know what happens if traitlets of the same name but different type are defined.

It would be possible to simplify the existing `Factory` implementation considerably, and fix this issue you have pointed out, if we don't rely on the `Factory` to replicate and pass the traitlets to the Components. Instead, one would need to directly specify the traitlets, identically to how its now done in #915.

I am still really in favour of keeping the Factory class separate to a Component. I don't think every Component should be automatically be given Factory behaviour...

MaxNoe, 2019-02-27:
#916 got merged, so closing this

closed 2019-02-27 by MaxNoe