cta-observatory/ctapipe #922: move event.mcheader or replace it with a metadata dict

kosack, 2019-01-10:
Right now there is a lot of MC header info in a `Container` (MCHeaderContainer) in the event data structure.  Two things should be done:

1) move this outside of the event data structure, since it does not change per event (similar to the `inst` container, which will also move soon

2) decide whether it makes sense that this info is in a Container at all: Containers represent rows in an output table, so if there is no use-case for writing this as a row,  we could just directly copy *all* header info in a simtel file into e.g. the `event.mc.meta`   dictionary (the metadata for the event.mc table), which would be output as attributes or header keywords in the output file by a TableWriter subclass.

kosack, 2019-01-10:
It was noted in #921 that there is a use case for (2) being a `Container`, in that we might merge multiple simtel files into one output file, in which case we'd want a table of these parameters (and also perhaps that means we need to store the mc run number with each event to be able to associate the two)

maxnoe, 2020-10-26:
I think with the dl1 tool developement it is pretty clear that we want to have access to this kind of information in a container and that it makes sense that an event knows about its mc header, since we might work on an event stream from several runs.

closed 2020-10-26 by maxnoe