cta-observatory/ctapipe #938: pandas 0.24.0 breaks ctapipe

dneise, 2019-01-28:
## Reminder: 

As we can see in #937 pandas 0.24.0 broke the master. It works with pandas 0.23.4
We should not too long stick to an old version of pandas but rather find out what the problem is.

kosack, 2019-01-28:
From the pandas 0.24 docs:

>Warning pandas requires PyTables >= 3.0.0. There is a indexing bug in PyTables < 3.2 which may appear when querying stores using an index. If you see a subset of results being returned, upgrade to PyTables >= 3.2. Stores created previously will need to be rewritten using the updated version.

kosack, 2019-01-28:
But we are using 3.3.4 of pytables, so this shouldn't be a problem...

kosack, 2019-01-28:
I was able to reproduce the problem locally:

The data are there:

```sh
> h5ls -r --full hillas.h5 
/                        Group
/dl1                     Group
/dl1/hillas              Dataset {70/Inf}
```

Reading as before fails (but this works in the previous version):

```py3
>>> pd.read_hdf("hillas.h5")                                               

ValueError: No dataset in HDF5 file.
```

But it does work if you specify the dataset:
```py3
>>> pd.read_hdf("hillas.h5", key='/dl1/hillas')          

       intensity   kurtosis    length  ...         width         x         y
0     516.973074   2.002860  0.241504  ...  4.722307e-02 -0.970985  0.385674
1   12351.140514   3.896901  0.257008  ...  1.143911e-01 -0.546333  0.627852
2     104.850078   5.350268  0.321235  ...  2.475438e-02  0.563663  0.584721
3     111.341088   2.111895  0.050280  ...  2.288178e-02  0.343397  0.897011
4      92.755466   5.135449  0.217182  ...  2.140994e-02 -0.470735  0.768036
5      50.130122   2.315119  0.202210  ...  1.669234e-02 -0.560238  0.622500
6     156.264453   5.088021  0.107545  ...  3.576984e-02 -1.010522  0.264090
7     276.659159   2.732760  0.061533  ...  4.703201e-02 -1.007649 -0.03
```

SO the problem is the feature where the docs say you can ignore the `key` option if there is only 1 dataset in the file. This behavior seems to have changed or broken. It's not critical, since you really should specify which dataset...

kosack, 2019-01-28:
In any case, we should update all examples to explicitly specify the key, since in most cases we will store more than one dataset in a file anyhow. But it does seem to be a bug in the latest pandas in the code that finds the first dataset.