cta-observatory/ctapipe #942: avoid duplicate docu? Or maybe cross-link?

dneise, 2019-01-29:
This is a question and maybe a reminder to myself for improving this. Not complaining at all.

I chose the example of `Tool` but there might be others. Currently we have 3 pages explaining what a Tool is and how to implement one:
 * [General introduction and development hints](https://cta-observatory.github.io/ctapipe/tools/index.html)
 * [doc-string of Tool, also containing an example of a Tool - outdated!](https://cta-observatory.github.io/ctapipe/api/ctapipe.core.Tool.html#ctapipe.core.Tool)
 * And newly we also have the `Tools.ipynb` which explains how to write a Tool and how to use it.

These places feel to me pretty disjointed, i.e. one developer might only ever see one of these pages and once they read and understood what it says, move on to complete the task at hand. In case these places would disagree with each other, we create a bunch of developers who disagree with each other while at the same time complying with what they believe is **the documentation**.

Also it feels somehow duplicated, and since keeping documentation from out-dating is a super-human task already, I'd say keeping duplicate docu from out-dating is almost impossible. 

However: 

1. documenting in doc-strings is good practice and well known. Developers are actually used to search for documentation in doc-strings. IDEs parse docstrings and show them while we are developing.
2. Examples in notebooks are just great. They look great and feel great.
3. Having a general introduction about a certain and important part of the framework, which is outside of the code seems very reasonable. Since it **is not clear** how Components, Containers and Tools play together without some general guidance.

MaxNoe, 2019-01-29:
Why is there a notebook on how to write an command line application? That sounds strange to me

kosack, 2019-01-29:
Now that the API docs are at a lower-level, we can start to separate them from user docs.  The main problem is that nobody has even touched the documentation in a long while. 

I think the best way would be:
* examples and tutorials are notebooks  as now (easy to edit, and ensure code runs correctly)
* the index.rst files for each API module should be turned into just a small boilerplate, rather than the complex mess we have now for some of them
* general info on how to use a module or sub-module could move into the docstring  of thue code itself (which gets automatically put into the docs if automodapi is used). That has the advantage that the low-level API docs are right in the code, so people remember to update them. 
* any other high-level documentation should be removed from the API docs, and put into a section called "users guide" (and crosslinked with tutorials and examples)

How does that sound?

dneise, 2019-01-30:
> I have had on my list of things to do (for a long time)

I personally think, instead of investing additional time here, one might already transport 95% of the information by simply putting a link to the presentation somewhere into the docks. Even if the link leads to the non-public CTA web-sites .. many of the potential readers will have access. 

On the other hand, assuming somebody who is just browsing the ctapipe sphinx docs, might also search for potentially interesting presentations on the non-public CTA websites **and** actually find them might be a bit far fetched.

closed 2025-08-21 by maxnoe