cta-observatory/ctapipe #950: ctapipe vs eventdisplay id

thomasgas, 2019-02-05:
I'm having a look at some files with ctapipe and a collegue is using EventDisplay. We are trying to see the differences in the cleaning and reconstruction event by event but the naming of the events are quite different.
so for me `event.count = 3`, while for her `Event = 1400`. 
for another event i have `event.count  = 11`, and she has `Event = 10200`. 

*Question*: how to we easily go from one event id to the other?

MaxNoe, 2019-02-05:
Have a look not into count, this just counts the triggered event, but into `event.r0.event_id` and `event.r0.obs_id`.

thomasgas, 2019-02-05:
thanks! 
event.r0.event_id and event.r0.obs_id are exactly what I was looking for.
Thanks @GernotMaier too!

closed 2019-02-05 by thomasgas