cta-observatory/ctapipe #952: Issue with installation of ctapipe

chaimain, 2019-02-05:
https://github.com/cta-observatory/ctapipe/blob/37f70ce678a43fc1ccc956572a25e5892e73a4a9/setup.py#L63

While installing ctapipe a check was conducted on the version of pyhessio even though the new pyeventio is used now, as per the FAQ.
I tried to install the latest version of pyhessio separately as per the instructions, given in https://cta-observatory.github.io/pyhessio/getting_started/index.html and that caused an error while doing the pytest - 
> _pytest.config.ConftestImportFailure: (local('local_address/ctapipe/conftest.py'), (<class 'ValueError'>, ValueError('numpy.ufunc size changed, may indicate binary incompatibility. Expected 216 from C header, got 192 from PyObject',), <traceback object at 0x7fec64c0ab48>)

I then removed the pyhessio package and edited the setup.py file of ctapipe to comment out the check for pyhessio, but I still receive this error which I am unable to discern why.
I have python 3.6.8 currently.

thomasgas, 2019-02-07:
Sorry for being late in replying to you.
> [ctapipe/setup.py](https://github.com/cta-observatory/ctapipe/blob/37f70ce678a43fc1ccc956572a25e5892e73a4a9/setup.py#L63)
> 
> Line 63 in [37f70ce](/cta-observatory/ctapipe/commit/37f70ce678a43fc1ccc956572a25e5892e73a4a9)
>  'pyhessio>=2.1', 
> 
> While installing ctapipe a check was conducted on the version of pyhessio even though the new pyeventio is used now, as per the FAQ.

I think this comes from some unit test used to make sure that the pyhessio and pyeventio gives the same output: in order to do this you still need to install pyhessio.

> I tried to install the latest version of pyhessio separately as per the instructions, given in https://cta-observatory.github.io/pyhessio/getting_started/index.html and that caused an error while [...]

Have you tried installing ctapipe using Anaconda or miniconda environments as suggested here https://cta-observatory.github.io/ctapipe/getting_started/index.html?

MaxNoe, 2019-02-07:
> _pytest.config.ConftestImportFailure: (local('local_address/ctapipe/conftest.py'), (<class 'ValueError'>, ValueError('numpy.ufunc size changed, may indicate binary incompatibility. Expected 216 from C header, got 192 from PyObject',), <traceback object at 0x7fec64c0ab48>)

This sounds like a compilation issue (different compiler vs. runtime). 

What is your setup? OS, python...

chaimain, 2019-02-07:
> Sorry for being late in replying to you.
>
>
> I think this comes from some unit test used to make sure that the pyhessio and pyeventio gives the same output: in order to do this you still need to [...]

Yes, I have used Anaconda environments as suggested by that page. I can try again to install pyhessio separately, but I don't know if the above mentioned error will go away.

chaimain, 2019-02-07:
> This sounds like a compilation issue (different compiler vs. runtime).
>
> What is your setup? OS, python...

I am using Ubuntu 16.04 and Python 3.6.8. What other information will be relevant to resolve this issue, if it is indeed a matter of compilation?

dneise, 2019-02-08:
Hello @chaimain 

I had a similar issue 2 weeks ago and must admit that I "solved" it by wiping my conda environment and re-installing it.

I remember we also had this issue once on our testing platform travis. There it was caused by conda installing a different python version after being halfway throught the installation process. This resulted then in some packages being compiled for python 3.7 and some being compiled for py3.6 or so. 

So I would propose as a test: 

 - remove your ctapipe conda env
 - reinstall it
 - copy the whole shell output into a text file and put it here into this issue. (just drag and drop)

dneise, 2019-02-08:
**Update:** 
I am again running now again also into this issue on my laptop...

dneise, 2019-02-08:
Okay @chaimain could you please try the following?
(make sure you are in your conda cta environment)

    pip uninstall eventio --yes
    pip install eventio==0.14

Your output might contain a warning like here:
```
$ pip install eventio==0.14
Collecting eventio==0.14
Requirement already satisfied: corsikaio~=0.2.0 in /home/dneise/anaconda3/envs/bazbaz/lib/python3.7/site-packages (from eventio==0.14) (0.2.0)
Requirement already satisfied: numpy in /home/dneise/anaconda3/envs/bazbaz/lib/python3.7/site-packages (from eventio==0.14) (1.15.4)
ctapipe 0.6.2.post2 has requirement eventio==0.11.0, but you'll have eventio 0.14.0 which is incompatible.
Installing collected packages: eventio
Successfully installed eventio-0.14.0
```

On my machine this fixed the issue.

chaimain, 2019-02-11:
> Okay @chaimain could you please try the following?
> (make sure you are in your conda cta environment)
> 
> ```
> pip uninstall eventio --yes
> pip install eventio==0.16     # or the newest  .. 0.16
> ```
> Your output might contain a warning like here:
> 
> ```
> $ pip install eventio==0.14
> Collecting eventio==0.14
> Requirement already satisfied: corsikaio~=0.2.0 in /home/dneise/anaconda3/envs/bazbaz/lib/python3.7/site-packages (from eventio==0.14) (0.2.0)
> Requirement already satisfied: numpy in /home/dneise/anaconda3/envs/bazbaz/lib/python3.7/site-packages (from eventio==0.14) (1.15.4)
> ctapipe 0.6.2.post2 has requirement eventio==0.11.0, but you'll have eventio 0.14.0 which is incompatible.
> Installing collected packages: eventio
> Successfully installed eventio-0.14.0
> ```
> On my machine this fixed the issue.

@dneise  Sorry for the late reply.
I tried reinstalling the conda env and eventio as you mentioned. It did give a similar warning as you mentioned, but it did not completely solved the issue. I got the following errors while performing pytest:
============================================================================== ERRORS ===============================================================================
______________________________________________________ ERROR collecting pyhessio/pyhessio/tests/test_hessio.py ______________________________________________________
../../anaconda2/envs/cta-dev/lib/python3.7/site-packages/py/_path/local.py:668: in pyimport
    __import__(modname)
pyhessio/pyhessio/__init__.py:107: in <module>
    lib_close = np.ctypeslib.load_library('pyhessioc', _path)
../../anaconda2/envs/cta-dev/lib/python3.7/site-packages/numpy/ctypeslib.py:155: in load_library
    raise OSError("no file with expected extension")
E   OSError: no file with expected extension
The test was interrupted with 4 skipped, 7 warnings and the above error in 6.14 seconds.

I had installed pyhessio with python 3.7 and numpy 1.16.1

dneise, 2019-02-12:
Hello @chaimain thanks for your patience.

To me this looks like an error with pyhessjoio. The previous error came from eventio. So you are already one step further now. I am no expert for installing pyhessio, but you could try to reinstall pyhessio like:

    # activate your cta environment before
    pip install https://github.com/cta-observatory/pyhessio/archive/v2.1.1.tar.gz

dneise, 2019-02-12:
Maybe we could speed up this issue with a shared screen meeting, like skype or similar?

chaimain, 2019-02-12:
> Maybe we could speed up this issue with a shared screen meeting, like skype or similar?

Yes, we can do it today. Is it ok to do it after 3 pm CET? I have Skype, Google Hangouts, or any other that is favorable, can be arranged.

dneise, 2019-02-12:
Oh sorry I was pretty busy... my skype is "dominik.neise"

dneise, 2019-02-19:
We met on skype and got it running, I think the problem was caused by outdated documentation. @chaimain had `pyhessio` and `ctapipe` in the same folder, so plain `pytest` wanted to test both projects... which is not a problem by itself, but he followed the installation instrauctions here, I believe:

https://github.com/cta-observatory/pyhessio#build-and-install

while this is mostly only needed for people who plan to develop something within pyhessio and not in ctapipe. 

But since we merged #957 this should not be a problem anymore. 

Closing this

closed 2019-02-19 by dneise