cta-observatory/ctapipe #958: Handling of traitlets config for subcomponents

MaxNoe, 2019-02-08:
Follow up to #947, which got merged before the discussion was finished.

The question was, if to call the traitlets related kwarg `parent` or `tool`. For now it's `tool`.

I think it should be `parent` and that we were using those arguments not corrrect e.g. in `CameraCalibrator`.

A tool or Component initialiazing components as members, should only pass itself as `parent` of that component.

See this example:
```
from traitlets.config import Configurable, Config
from traitlets import Float

class Child(Configurable):
    value = Float(default_value=3.0).tag(config=True)

class Parent(Configurable):

    def __init__(self, config=None, parent=None):
        super().__init__(config=config, parent=parent)
        self.child = Child(parent=self)

config = Config()
config.Child.value = 10.0
p = Parent(config=config)
print(p.child.value) # output: 10
```

dneise, 2019-02-08:
Max and me discussed this a while in a chat and on the phone...

For me, the important point here is, that I was entirely unaware of this feature of `traitlets.config.Configurable` namely that a Configurable will use the configration from its `parent` if no `config` is given.

I would also like to point out that `config` might get precedence over `parent` like here:
```python
from traitlets.config import Configurable, Config
from traitlets import Float

class Child(Configurable):
    value = Float(default_value=3.0).tag(config=True)

class Parent(Configurable):

    def __init__(self, config=None, parent=None):
        super().__init__(config=config, parent=parent)
        child_config = Config()
        child_config.Child.value = 50.
        self.child = Child(parent=self, config=child_config)

config = Config()
config.Child.value = 10.0
p = Parent(config=config)
print(p.child.value) # output: 50
```

but in case `config` is wrongly constructed like here, neither `parent` nor `config` is used, but `Child` falls back to its default.

```python
from traitlets.config import Configurable, Config
from traitlets import Float

class Child(Configurable):
    value = Float(default_value=3.0).tag(config=True)

class Parent(Configurable):

    def __init__(self, config=None, parent=None):
        super().__init__(config=config, parent=parent)
        child_config = Config()
        child_config.value = 50.  # note! "child" is missing!
        self.child = Child(parent=self, config=child_config)

config = Config()
config.Child.value = 10.0  # this is totally fine
p = Parent(config=config)
print(p.child.value) # output: 3. # still Child falls back to its own default
```

I would also like to point out that no warnings were created in this case whatsoever.

MaxNoe, 2019-02-08:
Seeing this, I would forbid it explicitly to procide both parent and config:
```
if parent is not None and config is not None: 
    raise ValueError("Only one of `config` or `parent` allowed')
```

dneise, 2019-02-08:
I will make a proposal, so we can look at it.

dneise, 2019-02-08:
> if parent is not None and config is not None: 
    raise ValueError("Only one of `config` or `parent` allowed')

When I try to build in this `if` into `Component` then this clashes with the current `Factory` in the master. 
The factory has some features, like building the `config` .. e.g. in this case:
```python
self.reader = EventSourceFactory.produce(parent=self)
```

The factory internally creates a `config` object and somewhere calls:
```
factory = cls(config=config, parent=parent, **kwargs)
```

Where `config` was built inside the factory and is "essentially empty" since all the config is inside the `parent`. So there is a line in factory where config is not None and parent is not None ...

dneise, 2019-02-08:
I will now look if this can better be implemented on top of the `modified_factories` branch then the current master ... maybe I am lucky.

dneise, 2019-02-08:
Yep **seems** to work ... have a look at #959.

MaxNoe, 2019-03-05:
Now in #998

closed 2019-05-22 by kosack