cta-observatory/ctapipe #960: ctapipe-info exception when dependency is too new

dneise, 2019-02-10:
I've got eventio `0.16` installed ctapipe wants `0.11` and just for fun I wanted to see how this is displayed in `ctapipe-info`.

Answer: with an exception:
```
08:09 $ ctapipe-info --all
Traceback (most recent call last):
  File "/home/dneise/anaconda3/envs/cta/lib/python3.7/site-packages/pkg_resources/__init__.py", line 581, in _build_master
    ws.require(__requires__)
  File "/home/dneise/anaconda3/envs/cta/lib/python3.7/site-packages/pkg_resources/__init__.py", line 898, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/home/dneise/anaconda3/envs/cta/lib/python3.7/site-packages/pkg_resources/__init__.py", line 789, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.ContextualVersionConflict: (eventio 0.16.0 (/home/dneise/cta/pyeventio), Requirement.parse('eventio==0.11.0'), {'ctapipe'})

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/dneise/anaconda3/envs/cta/bin/ctapipe-info", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/home/dneise/anaconda3/envs/cta/lib/python3.7/site-packages/pkg_resources/__init__.py", line 3126, in <module>
    @_call_aside
  File "/home/dneise/anaconda3/envs/cta/lib/python3.7/site-packages/pkg_resources/__init__.py", line 3110, in _call_aside
    f(*args, **kwargs)
  File "/home/dneise/anaconda3/envs/cta/lib/python3.7/site-packages/pkg_resources/__init__.py", line 3139, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/home/dneise/anaconda3/envs/cta/lib/python3.7/site-packages/pkg_resources/__init__.py", line 583, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/home/dneise/anaconda3/envs/cta/lib/python3.7/site-packages/pkg_resources/__init__.py", line 596, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/home/dneise/anaconda3/envs/cta/lib/python3.7/site-packages/pkg_resources/__init__.py", line 784, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'eventio==0.11.0' distribution was not found and is required by ctapipe
```

I think this should not be the case.

MaxNoe, 2019-02-10:
This does not home from ctapipe Ordner, nur is python's behaviour.

As you can see from the trace back, we never even entered ctapipe code. 

The only solution for this would be to drop the exact requirement on the pyeventio version.

I think this behaviour is the only valid one, if we require an exact version: fail early, fail loudly

dneise, 2019-02-12:
Ok

closed 2019-02-12 by dneise