cta-observatory/ctapipe #963: Robust neighbour calculation

watsonjj, 2019-02-11:
The pixel neighbour calculation may need revisiting. Currently it defines neighbours based on pixel separation. Right now, the defined separation seems to result in a different number of neighbours for different pixels on the CHEC-S camera. From examples/display_integrator.py:

[image]

I did not expect more than four neighbours for each pixel. Perhaps a more robust definition of a pixel neighbour should be implemented. Or perhaps it is fine to have N pixels within a separation to be defined as a neighbour. I will look into it.

watsonjj, 2019-03-15:
This occurs due to the curved focal plane of the camera, which results in a changing pixel seperation with distance from the camera centre. The current neighbour definition does not account for this.

MaxNoe, 2019-03-18:
I cannot reproduce this locally with the CHEC camera geometry that comes with ctapipe, can you provide a file where this happens?

MaxNoe, 2019-03-18:
[image]

watsonjj, 2019-03-18:
Yeh sure. It occurs for the new CHEC geometry in Prod 4. An example simtel file can be obtained from this link:
https://www.dropbox.com/s/dlb05ntljpjhfil/run1.simtel.gz?dl=0

MaxNoe, 2019-03-18:
My robust method results in this for the CHEC camera:

[image]

Is this wanted?

watsonjj, 2019-03-18:
No, usually we do count cross-module pixels as neighbours

MaxNoe, 2019-03-18:
If I increase the tolerance to 0.35 I get the "correct" neighbors.

It's important that this threshold is smaller than sqrt(2) to not count the diagonals as pixels.

On the master it was `0.4`, but only evaluated on the central pixel, not for each pixel. This is why it broke for curved sensor plane.

watsonjj, 2019-03-18:
I am happy to take a look at your alternative solution. I took the approach of defining the pixels in terms of row and columns, as that tends to be how we do this sort of thing in the CHEC group. The utility function for obtaining row and column could be useful to have (perhaps adapted to be more robust in a similar fashion as you have done for the neighbours).

kosack, 2019-03-19:
> usually we do count cross-module pixels as neighbours

is there a reason for that? It's not clear to me if it is better or worse to count them. Has a study been done? Should be easy to check the resulting Hillas parameter differences.

watsonjj, 2019-03-27:
Fixed by #1018

closed 2019-03-27 by watsonjj