cta-observatory/ctapipe #964: LST reader update

FrancaCassol, 2019-02-11:
Hi,

I would like to make a PR  for updating the LST reader. 
Some weeks ago, It was said that the camera readers should go out from ctapipe and be modified as plugin. Is this development foreseen in a short time  or can I go ahead with the PR for updating the reader?

Thanks

Franca

MaxNoe, 2019-02-27:
The plugin system got merged and @dneise made a repository for the sst1m event source. LST reader could do the same.

MaxNoe, 2019-02-27:
I extracted the relevant source code into this repo:
https://github.com/cta-observatory/ctapipe_io_lst

closed 2019-02-27 by MaxNoe

FrancaCassol, 2019-02-28:
Thanks Max,

I will try to extract it  and then move it out of ctapipe

MaxNoe, 2019-02-28:
@FrancaCassol I don't understand. I already did. This is what is in the repository I linked. And I also opened a ctapipe PR that removes it here #992 

I will now do the same for the nectarcam.

MaxNoe, 2019-02-28:
Same for NectarCam here https://github.com/cta-observatory/ctapipe_io_nectarcam and https://github.com/cta-observatory/ctapipe/pull/993

FrancaCassol, 2019-02-28:
> @FrancaCassol I don't understand. I already did. This is what is in the repository I linked. And I also opened a ctapipe PR that removes it here #992

Ok! I thought you were showing me an example :-)

Perfect. I will do a PR to update the LST reader in this new context, that in ctapipe is outdated
Thanks for the help