cta-observatory/ctapipe #968: Camera frame transformations

watsonjj, 2019-02-15:
When plotting star positions on the camera (transforming from sky coordinates to the camera frame), the positions do not appear where one would expect for different telescope types:

LST
[image]

CHEC
[image]

I would expect to see a mirroring in the second image with respect the the first, due to the 2M telescope.

As I understand now (from talking to @ParsonsRD), the camera frame is not the actual coordinates you expect to have when physically looking at the face of the camera, but it is instead a "MCCameraFrame". This frame has already taken into account this reflection in the simulations, and therefore does not need to be handled differently in the analysis.

However, when handling observations from real camera data, this approach is not intuitive.

@ParsonsRD has suggested to include an additional camera frame which represents what you would see if you look at the face of the camera:

CameraFrame (New frame which represents what you see when you look at the camera face. Data read from real telescopes are associated with this frame)
    &#8595;
MCCameraFrame (Currently existing CameraFrame. Data read from simulations are associated with this frame)
    &#8595;
TelescopeFrame, etc.

watsonjj, 2019-02-15:
Code snippet for the above plots:

```python
import matplotlib.pyplot as plt
from astropy.time import Time
from astropy.coordinates import SkyCoord, EarthLocation
import astropy.units as u
from ctapipe.coordinates import CameraFrame, HorizonFrame
from ctapipe.visualization import CameraDisplay
from ctapipe.instrument import CameraGeometry

location = EarthLocation.of_site('Roque de los Muchachos')
obstime = Time('2018-11-01T04:00')

crab = SkyCoord.from_name("crab nebula")

horizon_frame = HorizonFrame(location=location, obstime=obstime)
pointing = crab.transform_to(horizon_frame)
camera_frame = CameraFrame(
    telescope_pointing=pointing,
    focal_length=u.Quantity(2.15, u.m),#28 for LST
    obstime=obstime,
    location=location,
)

# cam = CameraGeometry.from_name('LSTCam')
cam = CameraGeometry.from_name('CHEC')
fig, ax = plt.subplots()
display = CameraDisplay(cam, ax=ax)

ax.set_title(
    f'La Palma, {obstime}, az={pointing.az.deg:.1f}°, zenith={pointing.zen.deg:.1f}°'
)

stars = [crab]
for name in ['o tau', 'zet tau']:
    stars.append(SkyCoord.from_name(name))
for star in enumerate(stars):
    
    star_cam = star.transform_to(camera_frame)

    x = star_cam.x.to_value(u.m)
    y = star_cam.y.to_value(u.m)

    ax.plot(x, y, marker='*', color=f'C{i}')
#     ax.annotate(
#         text=name, xy=(x, y), xytext=(5, 5),
#         textcoords='offset points', color=f'C{i}'
#     )

# plt.show()
```

watsonjj, 2019-02-20:
Connected to #255

thomasgas, 2019-02-21:
I agree to have this additional Frame: if this is added, I can also add it to the coordinate notebook which I am preparing.

I would also add this frame for LSTCam and the other cameras, since know the plotting of the camera is not realistic. You can compare what I produced in #875 with what an observer should see when really looking at the camera (have a look at https://forge.in2p3.fr/projects/cta_analysis-and-simulations/repository/show/Simulations/MCModelDescription/trunk/reports)

ParsonsRD, 2019-02-21:
Agreed I would propose to place this frame in between the CameraFrame and the TelescopeFrame, such that the MC can be read into the MCCamera frame, but transformed back to the "real" camera frame if needed.

watsonjj, 2019-10-14:
Fixed by #1085

closed 2019-10-14 by watsonjj