cta-observatory/ctapipe #977: Implement a standard analysis

MaxNoe, 2019-02-20:
As discussed in Dortmund

- [ ] A Tool for DL0 → DL1
- [ ] A Tool for splitting data into training / test sets
- [ ] A Tool for training models
- [ ] A Tools for DL1 → DL2 applying those models
- [ ] A Tool creating irfs from DL2 test sets
- [ ] A Tool creating DL3 from irfs and DL2

Anything wrong /  missing?

vuillaut, 2019-02-20:
I would already add a tool for handling configuration.

watsonjj, 2019-03-07:
Can I confirm that the intention is to have a full pipeline to DL3 (and the methods involved) contained inside ctapipe? I am very conscious about the different IRF and reconstruction packages that exist across Github, and it would be good if we agreed on the approach we want to take for this. I personally would like to see the methods from these various packages implemented into ctapipe.

watsonjj, 2019-03-07:
Additionally, I would like to work on the DL0 -> DL1 tool. There are improvements to the calibration chain that have been on my to-do list for a long time now (and exist in various issues on this repository), which I hope to address soon.

kosack, 2019-04-12:
For this issue, I think it will help a lot when the protopipe code is released (which is now going to be before the Lugano CTA meeting).  It is code based on an older version of ctapipe, and doesn't use some ctapipe features (Tool, EventWriter, etc), but it implements a stable reference pipeline. 

I think the first work to do once it is released, is to start to use it as a model to make a more ctapipe-compliant version that we can include in ctapipe itself (e.g. tools for DL1, DL2, DL3 production + training tools that all use standard Containers and the ctapipe config system, rather than a custom one).  Then we can slowly deprecate protopipe in favor of this standard version.