cta-observatory/ctapipe #979: maybe use nbstripout?

dneise, 2019-02-21:
Hey, 

just a question. There is still little tool: nbstripout
https://github.com/kynan/nbstripout

which strips the output from ipy-notebooks before git commits, if you like. It can also install itself nicely as a git "filter", so it strips the output automatically before git add, so this cannot be forgotten so easily. 

I played around with it .. and it looks kind of okay. Just wanted to know if others know other tools and if we would like to promote the usage of such a tool among fellow developers? maybe mention it in the docu, if we like it... or so.

thomasgas, 2019-02-25:
Seems a nice tool! 
This would be an alternative to the solution used in gammapy for uploading notebooks without the output, right?

kosack, 2023-03-02:
I'd like to bump this issue: we should use it as a pre-commit hook even.  I already had an issue in #2266 where some extra metadata was written the broke the documention build, even after removing all output.

I notice that by default it does not strip some metadata like `kernelspec`, which gets written when I use a multi-kernel jupyterlab instance.  So it also needs a config file (and seems to use [git's configuration files as shown here](https://github.com/kynan/nbstripout#stripping-metadata))

maxnoe, 2023-06-30:
We should go for one of the text notebook formats provided by jupytext, see PR #2221 

This would also address the issue here.

maxnoe, 2023-10-21:
This is addressed / no longer relevant since we switched to the text format for the docs notebooks

closed 2023-10-21 by maxnoe