cta-observatory/ctapipe #980: docstrings format

thomasgas, 2019-02-21:
A really small question: which format do we use for the docstrings in ctapipe?
I saw different usages in ctapipe, but maybe we should agree on one format and use this across the whole project, plus add this information in the documentation "for developers".

watsonjj, 2019-02-21:
It was in my understanding that we use numpy docstrings. Its certainly the format I have seen most of.

dneise, 2019-02-21:
Since we do not run tests for our doc strings as of now, I would propose to at least use some kind of linter to check ourselves. 

I found this SO answer, saying that `pydocstyle` can check for us, if our doc-strings follow numpydoc style.
https://stackoverflow.com/a/43124848

Would be nice to let this run on every new PR .. but until we can set that up, I would invite every developer to use that tool on their local machine and play around with it for a while.

watsonjj, 2019-02-21:
> Since we do not run tests for our doc strings

However, I did notice in my recent PRs that travis would fail if docstring code was not fully executable

MaxNoe, 2019-02-21:
> It was in my understanding that we use numpy docstrings. Its certainly the format I have seen most of.
https://cta-observatory.github.io/ctapipe/development/style-guide.html

MaxNoe, 2019-02-21:
> However, I did notice in my recent PRs that travis would fail if docstring code was not fully executable

Yes, this is becose code snippets are executed when building the docs

dneise, 2019-02-21:
That is a sphinx feature I assume?

MaxNoe, 2019-02-21:
yes. These `.. example` or `.. code` blocks

dneise, 2019-02-21:
So: `pydocstyle --convention=numpy ctapipe` finds about 1800 issues with the master ... so saying that so far we followed numpydoc style is maybe a bit of a stretch ... :-D .. but that should not prevent us from improving the current situation.

thomasgas, 2019-02-21:
> https://cta-observatory.github.io/ctapipe/development/style-guide.html

Ah shame on me...we already had this in the documentation :sweat_smile: 

But, as @dneise showed, we have room for improvement ;)

closed 2019-02-21 by thomasgas