cta-observatory/ctapipe #983: Allowed mirror types overly restrictive

MaxNoe, 2019-02-22:
`OpticsDescription` raises errors, if mirror type is not one of `DC` or `SC`.

This is overly restrictive, as there are many more mirror types, even within CTA, I think the LST for example is a segmented parabolic mirror, not a Davies Cotton. 

FACT for example is a hybrid, in between Davies--Cotton and parabolic.

For what is is type actually used? I did not find anything.

ParsonsRD, 2019-02-22:
Hi, yeah to be honest this information is not super useful, the optical configuration should not make any difference to our analysis. What would be useful though is just to store whether the telescope is single mirror or dual mirror. As if we introduce an "engineering camera frame" or something like that we will need to perform a flip of the camera for dual mirror optics at some point.

MaxNoe, 2019-02-22:
To be really useful (I think the muon analysis needs it @AMWMitchell, @rlopezcoto),
one would also need information about the single tiles positions and shadowing, right?

MaxNoe, 2019-02-22:
I started refactoring this in #982

GernotMaier, 2019-02-27:
Completely irrelevant side note: the MST is also not a DC, but of 'intermediate' design with the intention to improve the timing compared to a pure DC design...

MaxNoe, 2019-03-05:
Fixed in #982

closed 2019-03-05 by MaxNoe