cta-observatory/ctapipe #984: change in 3D Reco

thomasgas, 2019-02-25:
I'm working on fixing the HillasReconstructor class to work in divergent pointing (#873).
What do you think about changing this:
https://github.com/cta-observatory/ctapipe/blob/ef14205fa9f7be7028839d50b335eb9be1b16092/ctapipe/reco/HillasReconstructor.py#L108
to this:
```python 
def predict(self, hillas_dict, inst,  array_pointing, telescopes_pointings=None):
    ...
    if telescopes_pointings is None:
        telescopes_pointings = {tel_id: array_pointing for tel_id in hillas_dict.keys()}
```
In this way one has to pass the array_pointing when dealing with normal pointing, plus a dictionary of telescope's pointings if in divergent pointing mode. If we are in parallel pointing mode, then the `telescopes_pointings` can be built as shown above, so that this dictionary is passed around.

In my opinion, it more clear to pass pointings (as `SkyCoord` in the `AltAz` frame), with respect to one dictionary for the altitude and one for the azimuth.

MaxNoe, 2019-02-25:
:+1: for giving a mapping `telescope_id → SkyCoord`

thomasgas, 2019-02-25:
working on this in #946

MaxNoe, 2019-02-25:
Could this be placed into the pointing container when reading?

thomasgas, 2019-02-25:
I think `event.pointing` should be the correct place to store this info (there are many issues related to the pointing btw). 
@MaxNoe I saw in #807 that you were working on this, is it true?

maxnoe, 2021-05-05:
Fixed in #946

closed 2021-05-05 by maxnoe