cta-observatory/ctapipe #985: Reco method cleaning

thomasgas, 2019-02-25:
I've an idea to clean a bit some stuff around. Each item is connected to the one above.

- [ ] Show 3D reconstruction method is better than 2D method (parallel pointing)
- [ ] Make the 3D methos work also in divergent mode.
- [ ] Drop the 2D method. It also lacks of tests, so if we drop it we don't have to write them.

something else?

watsonjj, 2019-02-25:
> Drop the 2D method. It also lacks of tests, so if we drop it we don't have to write them.

It seems a shame to delete it entirely. It could be nice to have an implementation of every reconstruction method available in ctapipe... could just be deprecated for the time being, until someone revitalises it?

maxnoe, 2021-12-06:
We added tests for the hillas intersection. It seems to perform slightly worse, but is now tested and there is no real reason to remove it.

closed 2021-12-06 by maxnoe