cta-observatory/ctapipe #988: Add tests to replace tests that tested for magic numbers

MaxNoe, 2019-02-27:
In #982 I removed tests, that tested on specific magic numbers of e.g. pixel values in an old prod2 file,
affected were:

*  ctapipe/calib/camera/tests/test_calibrator.py
*  ctapipe/calib/camera/tests/test_dl0.py
*  ctapipe/calib/camera/tests/test_dl1.py
*  ctapipe/calib/camera/tests/test_r1.py
*  ctapipe/image/tests/test_charge_extraction.py
*  ctapipe/image/tests/test_waveform_cleaning.py
*  ctapipe/io/tests/test_simteleventsource.py
*  ctapipe/tools/display_summed_images.py

We should add tests, that test the use method on known, simple data structures.

E.g. for pulse integration, use a gaussian pulse and test if the correct values come out.

MaxNoe, 2020-05-15:
I think this was fixed largely by @watsonjj in his rewrite of the calibration.

Could you comment if this is solved?

watsonjj, 2020-05-15:
Yes I believe so

closed 2020-05-15 by watsonjj