cta-observatory/ctapipe #990: event_source() not compatible with prototype readers?

FrancaCassol, 2019-02-27:
Hi,

I see that that in the master all "Factories" disappeared. I admit I do not see the immediate improvement, but this is surely my limit.

Anyway, I understand that for reading events I must now use "event_source()" like 
----
from ctapipe.io import event_source
source = event_source("/ctadata/franca/LST/LST-1.7.Run00000.0529.fits.fz", max_events=None)
----

but it seems to recognise only MC data: 

----
ValueError: Cannot find compatible EventSource for 
	url:/ctadata/franca/LST/LST-1.7.Run00000.0529.fits.fz
in available EventSources:
	['SimTelEventSource']
----

is this correct?

MaxNoe, 2019-02-27:
I think this has more with the creation of the plugin architecture than with the removal of factories.

The other eventsource are not imported by default.

Could you try adding this line to your file:
```
from ctapipe.io.lsteventsource import LSTEventSource
``` 

I will commit a fix for the moment, until the eventsource is in a plugin

MaxNoe, 2019-02-27:
See here: https://github.com/cta-observatory/ctapipe/pull/991

kosack, 2019-02-27:
For more info:

The factories have changed - you now can use `EventSource.from_url(filename)` or you can use the `event_source()` helper function, if you don't need the Tool/config functionality.  I think now you have to install the non-MC event sources separately... I haven't tried that yet, but perhaps Dominik could help? He implemented that.  For now they must be installed with pip, since I haven't made conda packages

Basically there is still "Factory" functionality, but it's built-in to the base class, like other Python classes (similar to how astropy works as well), but the second change is that we don't bundle all the readers with ctapipe, since it was getting too hard to maintain and test
Because of this big API change, the next revision of ctapipe will be 0.7 (0.6.x still uses the old way)
I'm working now on releaseing 0.6.2 (it's turning out to be a bit complicated), but should be there soon

FrancaCassol, 2019-02-27:
Thanks Max, adding the line it works 

I related it to the Factories because before I was using the EventSourceFactory that I can not more do now. 

> See here: #991
Wonderful!

I am ready to put the reader in a plugin, but how does it works? 
Dominik, could you please indicate me an example?

Thanks

FrancaCassol, 2019-02-27:
> Because of this big API change, the next revision of ctapipe will be 0.7 (0.6.x still uses the old way)
> I'm working now on releaseing 0.6.2 (it's turning out to be a bit complicated), but should be there soon

Ok Karl, I see

Thanks

MaxNoe, 2019-02-27:
I can do this for all the event sources trying to keep the git history.

MaxNoe, 2019-02-27:
Basically we only need a python module that is called `ctapipe_io_xxxx`

PR #991 (2019-02-27): Add missing imports for event sources
Fixes #990
closed 2019-02-27 by mackaiver