cta-observatory/ctapipe #1000: Container Serialization - Deserialization

dneise, 2019-03-05:
Hello, some people would like to serialize containers ans transmit them over the wire ... e.g. using zmq .. and deserialize again into containers on the other side. 

I think this does not exist yet, right? @MaxNoe  I think you wrote `Container.asdict()`  which is already a huge step towards serialization, I think. I am currently looking into serialization into msgpack and back again. Seems to work quite nicely... 

As the msgpack people show here: https://github.com/msgpack/msgpack-python#packingunpacking-of-custom-data-type

For serializing custom types, one needs to write these encode/decode functions. custom types includes of course numpy arrays and scalars, as well as `astropy.Time` and the `SubarrayDescription` as well as Quantities...

I do not know if serializing to msgpack or to JSON or to BSON or to googles protobuffers is best or so.. and I also do not care, but I see that people want some kind of serialization. So I am looking into this.

I just wanted to open this issue, to inform you and also ask, if there is already some "standard" to follow or some helpers I am not aware of... E.g. `Container.from_dict()` would be nice...

MaxNoe, 2019-03-05:
Containers are pickable, so using pyzmq you can do `socket.send_pyobj(container)`

dneise, 2019-03-05:
Ah right! thanks ... didn't think of pickle at all!

MaxNoe, 2019-03-05:
That being said, it might not be the best approach, bet surely the fastest if both the sending and receiving end are python. 
json, bson, msgpack or the like would be great for inter-language communication.

MaxNoe, 2019-03-05:
> `Container.from_dict()`

Thats basically `Container(**dict)` right?

dneise, 2019-03-05:
Okay so since I looked into msgpack now for an hour or so ... I'll open a little PR for what I have so far.

dneise, 2019-03-05:
> Thats basically Container(**dict) right?

Also that I was not aware of. ... why was I not aware that containers can be constructed from dicts? 

Ah maybe since this is not how we work with them. Instead we construct them once and then "fill" them ...

dneise, 2019-03-05:
regarding inter-language ... so for msgpack one needs these custom encoders and decoders ... which will need to be re-implemented for every language ... but they are fairly simple and straight forward, I think.

MaxNoe, 2019-03-05:
> Also that I was not aware of. ... why was I not aware that containers can be constructed from dicts?

That's nothing special about containers, that's python...

dneise, 2019-03-05:
> That's nothing special about containers, that's python...

I disagree ... If one does not know `Container.__init__` by heart, and only ever sees this usage:

    data = SomeContainer()
    data.foo = 'bar'

One might simply assume `Container.__init__` to be empty or so, but it is not, the relevant part (which I just looked at) is:
```python
    def __init__(self, **fields):
        for k, v in fields.items():
            setattr(self, k, v)
```

dneise, 2019-03-05:
>>    Container.from_dict()

>Thats basically Container(**dict) right?

Also it only works if not `recursive=True` was used.

MaxNoe, 2019-03-05:
> Also it only works if not recursive=True was used.

Thats someting different, now you are talking about that `containter == container(**container.to_dict(recursive=True))`, which would be useful, I think.

> 
```
data = SomeContainer()
data.foo = 'bar'
```

Really? That is not how it is intended. See for example `hillas_parameters`:

https://github.com/cta-observatory/ctapipe/blob/423e61e15d2448e04334c33055bb226c454f4d09/ctapipe/image/hillas.py#L155-L166

MaxNoe, 2019-03-05:
> I disagree ... If one does not know Container.__init__ by heart, and only ever sees this usage:

Container is basically meant to be a dict with attribute access and predefined fields. 
Not allowing keyword arguments in the `__init__` would maximum surprise.

closed 2020-10-26 by maxnoe