cta-observatory/ctapipe #1003: writing traitlets.config.loader.Config to hdf5 file?

FrancaCassol, 2019-03-06:
Hi,
for the moment the monitoring containers (in ctapipe_io_lst/containers.py) are written to hdf5 files with the HDF5TableWriter class. I would like to add to these containers  a Field with the traitlets.config.loader.Config used to run the code in order to keep track of it. 
At a first look this seems not to work. Has somebody a suggestion on how to easily write this information to hdf5 files?

Thanks

dneise, 2019-03-06:
I think the config is stored automatically for you by the `Tool` to the `Provenance` of the tool. 
Some information about Provenance can e.g. be found here:
https://cta-observatory.github.io/ctapipe/examples/provenance.html

I do not think Provenance is already automatically written anywhere.

I think, if you want the config to be stored inside the HDF5 file, then the easiest would be to store the config inside the `.meta` dictionary of your output container. Everything which is stored in the `YourContainer.meta` dict will be stored in the HDF5 files header. 

Keep in mind, the meta stuff is supposed to be constant for many events and does not change from event to event. And since the config is also constant for many events in one call to your tool, I assume, it would not be wise to store this constant information in a containers `Field`, since then the config would be stored again and again for every event, which would create huge output files for no reason.

An example of how other people use this `.meta` dictionary of Containers you might want to have a look here:
https://github.com/cta-observatory/ctapipe/search?q=.meta%5B&unscoped_q=.meta%5B

FrancaCassol, 2019-03-07:
Hi Dominik,

> I do not think Provenance is already automatically written anywhere.
> 
looking at here
https://cta-observatory.github.io/ctapipe/examples/provenance.html#Storing-provenance-info-in-output-files 

it seems that provenance can be written as a HDF5 file header, but I admit not to understand how (not really expert of HDF5). It would indeed nice to have everything written, not only the config.

> I think, if you want the config to be stored inside the HDF5 file, then the easiest would be to store the config inside the `.meta` dictionary of your output container. Everything which is stored in the `YourContainer.meta` dict will be stored in the HDF5 files header.

Thanks for the suggestion, but I can not do    data.meta['config]= myconfig  isn't it? I must separate all the keys?  Also, how can I then see the header of a HDF5 file?

FrancaCassol, 2019-03-07:
Sorry, I close this issue by mistake

MaxNoe, 2019-03-07:
h5 files can store meta data for pretty much every object, be it the file itself, groups, datasets or tables.

I don't know by heart, how pytables does it. pytables is used in ctaipe for the outputfiles.

h5py does it with the `attrs` attribute, which is basically a `dict`, very similar to a fits header, but without the fits restrictions.

```
import h5py

f = h5py.File('test.hdf5')
f.attrs['provenance'] = 'I created this file'
```

dneise, 2019-03-07:
> Sorry, I close this issue by mistake

You can just reopen it.

dneise, 2019-03-07:
Maybe this notebook might help? 

https://gist.github.com/dneise/539de6e18fad392eefae0caa7f561ba1

In this video I discuss it.

https://youtu.be/BS1SZfL0Yfw

I think while discussing it, I found a bug in the TableWriter ... but I am not sure... I will further look into it.

FrancaCassol, 2019-03-07:
Hi Diminik,

thanks very much. As usual your explanation was very clear and it works!
On the same way I thought I could write the full provenance info with:

mydata.meta['provenance']= Provenance().as_json(indent=3)

but it seems that at the tool.start level the provenance dictionary is still empty. Not too bad, for the moment the Config is the most important part 

Thanks again

dneise, 2019-03-07:
Ok so we close this issue .... :+1:

closed 2019-03-07 by dneise