cta-observatory/ctapipe #1004: HDF5TableWriter does not write complete DataContainer

dneise, 2019-03-08:
As one can see in this notebook (the warnings in Box [11] and below) the HDF5 TableWriter does not write DataContainers completely to disk. 

https://gist.github.com/dneise/539de6e18fad392eefae0caa7f561ba1

Reading the documentation about the TableWriter here: 
https://cta-observatory.github.io/ctapipe/ctapipe_api/io/index.html#serialization-of-containers

It sounds like writing **any container** should be possible. 

But looking at the example here:

https://cta-observatory.github.io/ctapipe/examples/table_writer_reader.html

There is no container beeing written to disk, which contains sub-containers.

And the test here:
https://github.com/cta-observatory/ctapipe/blob/master/ctapipe/io/tests/test_hdf5.py#L25

does write an `MCEventContainer` which contains a `Map(MCCameraEventContainer)` but the test only tests if the data can be read and written without an exception, but it does not test if the actual data, which is contained in the `MCEventContainer` can actually be read back.

----

I just looked at the file with pytables and found that only the "count" field was written to file. Nothing else.
What did I do wrong?

dneise, 2019-03-08:
Ah found it:

https://github.com/cta-observatory/ctapipe/blob/1f30bc0389ddff2b93b87d8e870fb55174da5a06/ctapipe/io/hdf5tableio.py#L31-L33

MaxNoe, 2019-03-08:
While I think the current behaviour is somehow documented, I also think this is a valid issue

dneise, 2019-03-08:
Ok yes, I think you are right. It is a pretty puzzling feature after all, and after I found the documentation again, I believe I remember Cyril also ran into this a few month ago already.

I propose to open an issue named: "Implement recursive Container HDF5 Table Writer" and maybe discuss a few design feature in that issue, before somebody jumps into actually implementing it.

Good?

MaxNoe, 2019-03-08:
Yes, sounds good.

dneise, 2019-03-08:
I close this in favor of #1005

closed 2021-05-25 by maxnoe