cta-observatory/ctapipe #1005: Implement recursive Container HDF5 Table Writer

dneise, 2019-03-08:
(Connected to #1004)

I think Max and me propose to implement a recursive HDF5 table writer. This issue here could serve for a discussion of some design ideas. (I simply assume the design is not trivial, otherwise such a recursive writer might have already been implemented some time ago ... :-))

I assume one would utilize the existing `HDF5TableWriter` for simplicity and call its `write()` for every sub-container recursively in order to write each sub-container into individual tables. This should already get all the data to disk "in some way" and the only thing left to do, would be to add some kind of meta/header information for the reader, so it is able to piece everything back together again. 

Looking at the current reader implementation, i.e. its `read()` method, the interface allows/requests the user [to explicitly specify the "group" and "table" name to be read as well as provide the Container class to be used](https://cta-observatory.github.io/ctapipe/examples/table_writer_reader.html#Reading-one-row-at-a-time:) as in:
```python
for data in h5_table.read(group_name, VariousTypesContainer()):
    # data is of type `VariousTypesContainer`
    # ...
```

I personally find this interface difficult to use, if it would be intended to be actually user facing, which is probably not is. I think it would be nice to:

 - be able to browse the contents of the HDF5 file, i.e. provide something like a `.keys()` method.
 - be able to read a table, without the need to know by heart the type of container which was stored inside that table in the first place.

---

So I would envision a possible interface of a this "RecursiveHDF5TableReader" like this:
```python
with RecursiveHDF5TableReader(path) as file:
    print(file.keys()) # basically a list of "paths" to tables or maybe just "names" 
```

Once one knows some keys one is interested in, this can be done:
```python
with RecursiveHDF5TableReader(path) as file:
    for data in file['some_key']:
        pass
```

----

Based on this small interface, the reader must be able to somehow learn about the container class it is supposed to fill with the data from the tables, and in case that container had a deep hierarchy, the reader needs to be able to piece the necessary tables together in the right order. So some kind of mapping needs to be stored in the file as well. The exact implementation of this, I would not want to specify at this point, but let be decided by the person, which is actually implementing it. 

Does this sound feasible or did I overlook something?

MaxNoe, 2019-03-08:
I think for this work, we probably need a verision of #445, which I first tried to do in #501, but which was discussed in Dortmund as not the way we want to go (e.g. not checking the types when setting members, but when coercing to output types).

We also need to think about how we want to store maps, e.g. of telesocpes. 
Or lists, e.g. of dl1 containers or dl2 containers (different reconstruction methods)

kosack, 2019-03-08:
> I assume one would utilize the existing HDF5TableWriter for simplicity and call its write() for every sub-container recursively

Yes, this was always my plan, so I fully support this! I don't think it should be too difficult.  

One question to think about (which may not be really critical) is if we implement other TableWriters (e.g. to FITS or other formats) if something like this would still work, but I think the solution would just be to have a mode where we store each table in a separate file with directories for sub-structure in that case, rather than try to have the recursive data somehow forced into a set of HDUs.  For HDF5 I think it'll work fine either way.

kosack, 2019-03-08:
One other question: how will the sub-structure be represented in HDF5 files? As flattened tables, as multiple tables, or as hierarchical columns in a single table? All are possible in HDF5, but it's not clear to me which is nicer.  Probably the latter breaks any possibility to read it in pandas, though, which might mean separate datasets is better. We could try to make the structure decision an option.  I would suggest these three possibilities for an output 'mode':

* **flat**: the output is a single flat table, where any sub-structure is flattened into multiple columns (e.g. you would get flat columns with the hierarchy encoded in their names like "events_tel002_hillas_x", "events_tel003_hillas_y"...). This means very large number of columns for deep datastructures, but is useful in cases where you want to read e.g. with pandas, etc. 
* **single**: a single dataset containing a table with HDF5 *hierarchical* columns 
* **multiple**: the structure is encoded in the hierarchy of datasets, and each dataset contains simple non-hierarchical tables. This is also probably the most efficient way to store sparse data. 

There may be others (e.g. different ways to flatten telescope_id maps)

dneise, 2019-03-08:
Ah, right. I somehow assumed Lists and Maps already worked. But if they do not yet work as you say. We have to make them work first. Right?

So... erm ... can we avoid inventing something and rather just take that functionality from somewhere? In #445 you make a point about the similarity of our Containers and ORMs. But if I remember correctly I have not yet seen a nice python ORM which transparently handles storing Llist-like or map-like fields for me.

I would really like to avoid inventing the wheel again here...

MaxNoe, 2019-03-08:
> Ah, right. I somehow assumed Lists and Maps already worked. But if they do not yet work as you say. We have to make them work first. Right?

I'm a bit confused. You opened this issue, because Maps were not written, right?

dneise, 2019-03-08:
I opened the issue because basically nothing inside a `DataContainer` was written.

MaxNoe, 2019-03-08:
Ok, so to clarify what works: only basic types like structure, int and float, arrays and quantities are written at the moment.

maxnoe, 2021-02-19:
I think we have this now with the `DL1Writer` and should open new issues when new things need to be added. 

Although a more general solution would also be great.

maxnoe, 2021-05-25:
This is no implemented using the `DataWriter`

closed 2021-05-25 by maxnoe