cta-observatory/ctapipe #1006: ArrayDisplay to be fixed

thomasgas, 2019-03-12:
While trying to analyse some data with a zenith of 60° for Paranal, I found a wrong behaviour in the ArrayDisplay:
[image]

Few considerations: 
- The reconstruction works (which is **good**) but the ArrayDisplay needs to be fixed:
- when you are in parallel mode, the plane of the camera and the TiltedGroundFrame are parallel and therefore you can use the angles obtained from the hillas parametrization to have the right plots, as shown here below: 

[image]

In this screenshot from my 3D displayer you see the frame, positions of the telescopes, the lines using the angles from the hillas parametrization and the impact point (both MC & reco) for each plane:
- in green (tilted frame)
- in red (ground frame)

The problem is only related to the plotting in the ground frame; we can't use the same angles to draw the "hillas_line" in the GroundFrame since angles are not preserved when doing projections and therefore have to be recomputed to have a right plotting. 

This was correct until now since we've always used MC with a zenith equal to 20°, for which the difference between the angles in the TiltedGroundFrame and the GroundFrame is less than a couple of degree.

I'm working now to modify the ArrayDisplay in order to have the right angles and solve this issue (and also https://github.com/cta-observatory/ctapipe/issues/737).

thomasgas, 2019-03-15:
I'm coming with a solution for this plotting. It works for Paranal, where the array is basically flat, 

[image]

But not so well for La Palma (especially at high zenith). Here is an image obtained with a custom MC production done at az=150° and zen=60° for La Palma :

[image]

In La Palma you can see that the angle is right now, but the line should be translated. The lines have the same z of the telescope whereas the impact point have z=0. 
So we can't just project them on the ground giving them a z=0 but they should be translated...it's difficult to explain 3d things just by writing :sweat_smile:

thomasgas, 2019-03-15:
So i reprojected the positions of the telescopes in the ground frame from the tilted frame in order to make the lines passing through that point. It works:
[image]

thomasgas, 2019-03-20:
Update: everything works in the tilted frame (reconstruction and ArrayDisplay):
```python
horizon_frame = AltAz()

array_pointing = SkyCoord(
    az=event.mc.az,
    alt=event.mc.alt, 
    frame=horizon_frame
)

fig, ax = plt.subplots(figsize=(10,10))

tilted_frame = TiltedGroundFrame(pointing_direction=array_pointing)

array = ArrayDisplay(
    event.inst.subarray.select_subarray(
        f'Paranal alt={array_pointing.alt:.2f}° az={array_pointing.az:.2f}°', hillas_dict.keys()
    ), 
    frame=tilted_frame,
    axes = ax)

#port the output of HillasReconstructor from Ground to Tilted
reco_core = GroundFrame(x=output.core_x, y = output.core_y, z = 0 * u.m)
reco_tilt = reco_core.transform_to(tilted_frame)
array.axes.scatter(reco_tilt.x, reco_tilt.y, c="r", label="reco", marker="+", s = 500)

#port the MC impact point from Ground to Tilted
mc_core = GroundFrame(x=event.mc.core_x, y = event.mc.core_y, z = 0 * u.m)
mc_tilt = mc_core.transform_to(tilted_frame)
array.axes.scatter(mc_tilt.x, mc_tilt.y, c="r", label="mc", marker="x", s = 500)

array.set_line_hillas(hillas_dict, range=1000)
array.add_labels()

array.axes.legend();
```
So that an event with 3 telescopes triggered gives this (in the tilted frame):
[image]
And this in the ground frame:
[image]

maxnoe, 2023-01-04:
@HealthyPear This is about the `psi_corr`, right? So it is fixed?

HealthyPear, 2023-01-06:
Yes, I think this was fixed during my tests with protopipe.

closed 2023-01-06 by maxnoe