cta-observatory/ctapipe #1007: SimTelEventSource does not have all telescopes for frankenstein simtel files

MaxNoe, 2019-03-14:
This is because eventio.SimTelFile stops reading objects in `__init__` as soon as it encounters the first `MCShower`, but in the case of a frankenstein file, the first shower might come in between the telescope descriptions.

thomasgas, 2019-03-14:
for the records: which are the frankenstein files?

kosack, 2019-03-14:
I guess those are ones where multiple telescope layout configurations were merged?

kosack, 2019-03-14:
It's probably also a problem in the old pyhessio, I guess - there if I remember correctly, we read the first event and then assume all headers have been filled at that point, but if some telescope headers come after, it would miss them.

kosack, 2019-03-14:
Options I see:
- Simply don't support those files (only needed now during certain studies), and emit a warning or error if a new telescope description is read after the first event
- scan the whole file first to find all headers, as in eventio_print_structure (effectively building an index table). However this  breaks the ability to "stream" or pipe these files, since you need to rewind. 
- have SimtelEventSource re-build the SubarrayDescription if it encounters a new header (and then any code using the SubarrayDescription needs to have it passed in explicitly for each event, and not use a cached copy)

MaxNoe, 2019-03-14:
> we read the first event and then assume all headers have been filled at that point, 

This is true when you read until the first *Array Event*, but currently we stop at the first *MC Event*

MaxNoe, 2019-03-14:
I implemented a fix in https://github.com/cta-observatory/pyeventio/pull/158

We initialize until we found all telescope descriptions.

MaxNoe, 2019-03-14:
> I guess those are ones where multiple telescope layout configurations were merged?

Exactly, this kind of Frankenstein file. There are others (multplile runs concatenated for example)

MaxNoe, 2019-03-14:
I released https://github.com/cta-observatory/pyeventio/releases/tag/v0.18.0

@kosack can you build the conda packages?

kosack, 2019-03-14:
The build is running now. Should have a package shortly.

kosack, 2019-03-14:
Ok, there should be eventio-0.18.0 packages for macOS & Linux for Py36 and Py37

MaxNoe, 2019-03-14:
https://github.com/cta-observatory/ctapipe/pull/1008/files

MaxNoe, 2020-04-21:
Fixed a long time ago

closed 2020-04-21 by MaxNoe