cta-observatory/ctapipe #1009: Remove EventSeeker

MaxNoe, 2019-03-14:
I don't think we should support this in the core package, it's basically a hack for event sources to provide random access, where that is not really necessary and quite expensive for most event sources we have.

kosack, 2019-03-14:
It does satisfy a few use cases, even if it's not something we encourage for production use. 

1. Debugging: we're actually right now trying to debug a reconstruction problem and being able to jump to a specific set of events by event_id (which is actually fairly fast if they are sorted first), makes this easier.  We could of course implement that differently, but this is pretty easy to use as is.

2. event GUI, where being able to go back and forth is useful, even if it is not particularly fast. 

Of course, if you are looking at DL1 or DL2 data, we get random access for free if they are stored in HDF5 or FITS tables, so having it in raw data is not completely critical since you could dump it first and then debug after, but why not?

watsonjj, 2019-03-14:
To add to Karl's comment, the ctapipe-event-viewer tool relies on the EventSeeker

watsonjj, 2020-07-23:
The EventSeeker still has uses in ctapipe and will be kept for now

closed 2020-07-23 by watsonjj