cta-observatory/ctapipe #1017: Fix broken tutorial and example notebooks after update of ctapipe-extra

kosack, 2019-03-19:
Some of the notebooks no longer do what they say due to changes in the gamma_test_large file in ctapipe-extra. They need some small cleanups. 

At least a few things I found

* pulse is in wrong place in "exploring raw data" , so now only noise is shown and integrated.  (easy to fix by choosing a new example event and changing a few parameters)
[image]

* strange rotation of FlashCam pixels in LST bootcamp (bug in CameraGeometry.pixel_rotation?)
[image]

kosack, 2019-03-19:
for the pixel rotation, it seems that either the data file has it wrong, or the SimtelEventSource is reading it wrong: `cam_rot=0.0 rad, pix_rot=0.0 deg`

MaxNoe, 2019-03-19:
This was discussed here https://github.com/cta-observatory/pyeventio/issues/144

The FlashCam was simulated with the wrong pixel orientation for most parts of PROD 3

kosack, 2019-03-19:
> The FlashCam was simulated with the wrong pixel orientation for most parts of PROD 3

Ok, not a real problem then. Do you know if there is any metadata in the simtel file that tells you the prod version? We could always hack a fix in the EventSource.  For now, perhaps just changing the example to another camera would be fine.

MaxNoe, 2019-03-19:
> We could always hack a fix in the EventSource

No, we cannot. The data was **simulated** with the wrong pixel orientation. E.g. photons were lost because of the gaps between the pixels. There is nothing we should fix. It is simply wrong in the file.

kosack, 2019-03-19:
Ok, more a reason to make some nice prod-4 example files then!

maxnoe, 2021-05-05:
I am closing this. If this is still relevant, feel free to reopen

closed 2021-05-05 by maxnoe