cta-observatory/ctapipe #1018: Proper handling of curved camera geometries

MaxNoe, 2019-03-20:
The information we need is in the SimTelFiles:

It contains pixel `x, y, z` and `x` and `y` of the normal vector on the pixel.
In the comment it says that the vector should be `(x, y, 1)`.

What is the angle to the z direction of the pixels of the CHEC camera @watsonjj ?
Just to verify.

MaxNoe, 2019-03-20:
```
s = SimTelFile('chec.simtel.gz')
camera_settings = s.telescope_descriptions[1]['camera_settings']

x = camera_settings['pixel_x']
y = camera_settings['pixel_y']
z = camera_settings['pixel_z']

# normal vector components
nx = camera_settings['nxpix']
ny = camera_settings['nypix']
nz = np.ones_like(nx)

```

The `chec.simtel.gz` is the file @watsonjj has provided here #963

MaxNoe, 2019-03-20:
@dneise maybe we should give better names to those `nxpix` and `nypix` in eventio

thomasarmstrong, 2019-03-20:
Looks sort of correct. The pixels should be orientated with each of the modules*, where the module positions can be found at start of the camera configuration file for chec. How do you only get 3 distinct pixel groups for the z angle? I get a range between 2 and 9 degrees.

*on a side note, is the corresponding module number for each pixel saved in camera_settings?

MaxNoe, 2019-03-20:
This is what I did:

```
from eventio import SimTelFile
import matplotlib.pyplot as plt
from mpl_toolkits.mplot3d import Axes3D
import numpy as np

s = SimTelFile('chec.simtel.gz')
camera_settings = s.telescope_descriptions[1]['camera_settings']

x = camera_settings['pixel_x']
y = camera_settings['pixel_y']
z = camera_settings['pixel_z']

# normal vector components
nx = camera_settings['nxpix']
ny = camera_settings['nypix']
nz = np.ones_like(nx)

n = np.column_stack([nx, ny])

print(*np.unique(np.round(np.rad2deg(np.arctan(n)), 1), axis=0), sep='\n')

fig = plt.figure()
ax = fig.add_subplot(1, 1, 1)

ax.quiver(x, y, nx, ny)
ax.set_aspect(1)

fig.savefig('chec.png', dpi=300)
plt.show()
```

The simtel files contains something called pixel drawers, maybe that's it?

MaxNoe, 2019-03-20:
[image]

thomasgas, 2019-08-21:
Can this issue be closed? @MaxNoe

MaxNoe, 2019-08-21:
No, we have no handling of curved geometries / the definite answer we don't need it

maxnoe, 2021-02-19:
I emailed with Konrad again, and he said we should like we do now just look at the xy projection. However that results in ugly camera displays, since there we don't actually do the projection but draw a regular square at each center position.

This also might create problems for the muon calibration or any other algorithm where you need to calculate a signal acceptance for each pixel. 

So input from the two mirror telescope teams here would be very welcome, I cannot yet wrap my head around how best to implement this. 

How do we need to transform a pixel with (x, y, n_x, n_y) where n_x and n_y are x and y components of the normal vector of the pixel to a position in the field of view (`TelescopeFrame`)