cta-observatory/ctapipe #1020: MCEventContainer, missing units and wrong primary ids?

MaxNoe, 2019-03-21:
* [ ] h_max and h_first_inter don't have units
* [ ] Primary ids in the comment are not the ones in corsika

See
https://github.com/cta-observatory/ctapipe/blob/cc215e2f7233973ff1b604f90c96ac8cea447fcd/ctapipe/io/containers.py#L235-L248

compare to CORSIKA:

[image]

kosack, 2019-03-21:
I think the primary_id is defined by sim_telarray, and isn't the same as corsika, but I may be wrong. 

For the h_max, I guess it's in m (deconvolved from the atmosphere model),  as opposed to x_max in g/cm^2.  Do we get both of those?

MaxNoe, 2019-03-21:
From `io_hess.h`:

```
   int primary_id;      ///< Particle ID of primary. Was in CORSIKA convention
                        ///< where detector_prog_vers in MC run header was 0,
                        ///< and is now 0 (gamma), 1(e-), 2(mu-), 100*A+Z
                        ///< for nucleons and nuclei, negative for antimatter.
```

MaxNoe, 2019-03-21:
What happens when I produce `e+` or `mu+`, protons are `101`?

closed 2021-02-19 by maxnoe