cta-observatory/ctapipe #1025: Some tools are broken

kosack, 2019-03-22:
Seems we are missing some tests, since the update to Tool and Component recently seems to have broken some tools.  For example:

```sh
$ ctapipe-reconstruct-muons --input=gamma_test_large.simtel.gz

INFO [MuonDisplayerTool] (tool/initialize): ctapipe version 0.5.2.post636+gitcc215e2
INFO [MuonDisplayerTool] (tool/run): Starting: ctapipe-reconstruct-muons
Traceback (most recent call last):
  File "/Users/kosack/miniconda3/envs/cta-dev/bin/ctapipe-reconstruct-muons", line 11, in <module>
    load_entry_point('ctapipe', 'console_scripts', 'ctapipe-reconstruct-muons')()
  File "/Users/kosack/Projects/CTA/Working/ctapipe/ctapipe/tools/muon_reconstruction.py", line 124, in main
    tool.run()
  File "/Users/kosack/Projects/CTA/Working/ctapipe/ctapipe/core/tool.py", line 165, in run
    self.setup()
  File "/Users/kosack/Projects/CTA/Working/ctapipe/ctapipe/tools/muon_reconstruction.py", line 71, in setup
    config=self.config, parent=self, eventsource=self.source
  File "/Users/kosack/Projects/CTA/Working/ctapipe/ctapipe/calib/camera/calibrator.py", line 78, in __init__
    super().__init__(config=config, parent=parent, **kwargs)
  File "/Users/kosack/Projects/CTA/Working/ctapipe/ctapipe/core/component.py", line 100, in __init__
    'Only one of `config` or `parent` allowed'
ValueError: Only one of `config` or `parent` allowed If you create a Component as part of another, give `parent=self` and not `config`
```

dneise, 2019-03-22:
There is a test, but it does not test the right thing... which might be a hint, that *all tool tests* do not tests the right thing:

Here is the test, for this tool:
https://github.com/cta-observatory/ctapipe/blob/cc215e2f7233973ff1b604f90c96ac8cea447fcd/ctapipe/tools/tests/test_tools.py#L12-L19

And here is what happens if one executes that test function body manually:
```python
In [9]: MuonDisplayerTool().run( 
   ...:     argv=shlex.split( 
   ...:         f'--events={GAMMA_TEST_LARGE} ' 
   ...:         '--max_events=2 ' 
   ...:     ) 
   ...: )                                                                                            
WARNING [MuonDisplayerTool] (loader/load_config): Unrecognized alias: '--events=/home/dneise/anaconda3/envs/cta/lib/python3.7/site-packages/ctapipe_resources/gamma_test_large.simtel.gz', it will probably have no effect.
INFO [MuonDisplayerTool] (tool/initialize): ctapipe version 0.6.2.post52+gitcc215e2
INFO [MuonDisplayerTool] (tool/run): Starting: ctapipe-reconstruct-muons
ERROR [MuonDisplayerTool] (tool/run): please specify --input <events file>.  Use --help for more info
```

---

So a couple of things are strange here, the test uses the `--events` cmd-line parameter, while Karl used `--input`. This is a mess.. gonna tidy that up.

dneise, 2019-03-22:
Okay so the problem is `tool.run()` never raises any exception. It swallows them all and prints them nicely to the cmdline. So there was no way to find out in the tests, if there was some kind of problem in `tool.run()`. 

I added a good ole integer return value now to `tool.run()` which can be used to at least find out if the tool had some kind of problem or not. This value should be checked in tools.

Another thing is the entry points for cmdline tools, currently they look like this:

    def main():
        tool = MuonDisplayerTool()
        tool.run()

But the **should** look like this (add a **return** statement)

    def main():
        tool = MuonDisplayerTool()
        return tool.run()

So that the shell, which invoked the tool has some chance to get its hands on these return codes. This is important for people using ctapipe tools within some bash scripts or so.

We should *require* that this `return` cannot be forgotten, with a bunch of tests, but I wonder if it is not easier, if we just automatically generate these entry points... somehow.

MaxNoe, 2019-03-22:
Shouldn't  it be sys.exit(tool.run)?

dneise, 2019-03-22:
Yep, that's happening, since these `entry_points['console_scripts']` which are created by `setup.py` do `sys.exit(whatever_is_returned_by_the_entrypoint)`.

MaxNoe, 2019-03-23:
Ah nice

dneise, 2019-03-25:
@cta-observatory/ctapipe-core-devs #1027 is ready to review

PR #1027 (2020-04-08): Fix tests for Tools, fixes #1214
Fixes #1025
closed 2020-04-08 by kosack