cta-observatory/ctapipe #1028: numpy.power(x, 2) --> x**2

thomasgas, 2019-03-25:
I happened to run some profiling on simple numpy functions:

```python
In [43]: x = np.random.rand(10)
In [44]: %timeit np.power(x,2)
1.94 µs ± 5.53 ns per loop (mean ± std. dev. of 7 runs, 1000000 loops each)
In [45]: %timeit x**2
1.05 µs ± 11.7 ns per loop (mean ± std. dev. of 7 runs, 1000000 loops each)

In [46]: x = np.random.rand(100)
In [47]: %timeit np.power(x,2)
2.62 µs ± 13.7 ns per loop (mean ± std. dev. of 7 runs, 100000 loops each)
In [48]: %timeit x**2
1.11 µs ± 7.62 ns per loop (mean ± std. dev. of 7 runs, 1000000 loops each)

In [49]: x = np.random.rand(1000)
In [50]: %timeit np.power(x,2)
9.97 µs ± 255 ns per loop (mean ± std. dev. of 7 runs, 100000 loops each)
In [51]: %timeit x**2
2.2 µs ± 9.53 ns per loop (mean ± std. dev. of 7 runs, 100000 loops each)

In [52]: x = np.random.rand(10000)
In [53]: %timeit np.power(x,2)
76.1 µs ± 383 ns per loop (mean ± std. dev. of 7 runs, 10000 loops each)
In [54]: %timeit x**2
4.77 µs ± 21.8 ns per loop (mean ± std. dev. of 7 runs, 100000 loops each)
 ```

So the better-looking `array**2` instead of `np.power(array, 2)` is from 2 to 30 times faster...we should change it in all the code.

This is valid only for `array**2`, whereas for `array**3` or the two methods have the same performance.

dneise, 2019-03-25:
Yes, I think we had this discussion already somewhere before. One can compare:
```
operator = x**2
numpy_power = np.power(x, 2)
inline = x * x
```

I believe `operator` was pretty much the winner.

watsonjj, 2019-04-12:
Is it possible to add custom rules to codacy, to check for things like this, and also the more efficient astropy units handling (#881)?

thomasgas, 2019-04-15:
i'm not sure codacy can also optimize code...maybe there's another plugin for this or we can build out own ones (don't know how).

watsonjj, 2019-04-15:
I meant can it flag whenever it literally sees `np.power(..., 2)`?

kosack, 2019-05-22:
Codacy isn't very configurable, perhaps once we have a CTA SonarQube instance we can have more specific guidelines like this implemented. 

Though of course, anywhere which such a low-level optimization is really needed should probably be use `@numba.njit`, in which case the numpy code is replaced with an optimized version anyhow, so I think we can generally ignore it.

commit e91e0ebb (2020-03-20): Replace np.power with power operator, fixes #1028 (#1239)

* Replace np.power with power operator, fixes #1028

* Fix refactoring typo

* Fix test after function rename

* Fix another renaming issue
closed 2020-03-20 by kosack