cta-observatory/ctapipe #1029: Configuration behaviour

watsonjj, 2019-03-25:
ChargeExtractor has no `window_width` traitlet (although some of its subclasses do). In our configuration system, one is allowed to configure this traitlet from the command line like so:

`ctapipe-display-integration -f /Users/Jason/Downloads/run1.simtel.gz --ChargeExtractor.window_width=1`

and like so:

```python
    from traitlets.config.loader import Config
    config = Config(
        {
            'ChargeExtractor': {
                'window_width': 20,
                'window_shift': 3,
            }
        }
    )

    local_peak_integrator = ChargeExtractor.from_name(
        'LocalPeakIntegrator',
        config=config,
    )
```

and like so, from a `Tool`:

```python
    aliases = Dict(
        dict(
            window_width='ChargeExtractor.window_width',
        )
    )
```
`ctapipe-display-integration -f /Users/Jason/Downloads/run1.simtel.gz --window_width 1`

__BUT__, if one does this last approach, then the following raises an error:
`ctapipe-display-integration -h`

```
Calibrate dl0 data to dl1, and plot the various camera images that characterise
the event and calibration. Also plot some examples of waveforms with the
integration window.

Options
-------

Arguments that take values are actually convenience aliases to full
Configurables, whose aliases are listed on the help line. For more information
on full configurables, see '--help-all'.

--id
    event_index will obtain an event using event_id instead of index.
Traceback (most recent call last):
  File "/Users/Jason/anaconda3/envs/cta/bin/ctapipe-display-integration", line 11, in <module>
    load_entry_point('ctapipe', 'console_scripts', 'ctapipe-display-integration')()
  File "/Users/Jason/Software/ctapipe/ctapipe/tools/display_integrator.py", line 358, in main
    exe.run()
  File "/Users/Jason/Software/ctapipe/ctapipe/core/tool.py", line 160, in run
    self.initialize(argv)
  File "/Users/Jason/Software/ctapipe/ctapipe/core/tool.py", line 123, in initialize
    self.parse_command_line(argv)
  File "<decorator-gen-4>", line 2, in parse_command_line
  File "/Users/Jason/anaconda3/envs/cta/lib/python3.7/site-packages/traitlets/config/application.py", line 87, in catch_config_error
    return method(app, *args, **kwargs)
  File "/Users/Jason/anaconda3/envs/cta/lib/python3.7/site-packages/traitlets/config/application.py", line 526, in parse_command_line
    self.print_help('--help-all' in interpreted_argv)
  File "/Users/Jason/anaconda3/envs/cta/lib/python3.7/site-packages/traitlets/config/application.py", line 386, in print_help
    self.print_options()
  File "/Users/Jason/anaconda3/envs/cta/lib/python3.7/site-packages/traitlets/config/application.py", line 357, in print_options
    self.print_alias_help()
  File "/Users/Jason/anaconda3/envs/cta/lib/python3.7/site-packages/traitlets/config/application.py", line 323, in print_alias_help
    trait = cls.class_traits(config=True)[traitname]
KeyError: 'window_width'
```

Therefore, one is forced to do workarounds like the abstract class `WindowIntegrator` and `PeakFindingIntegrator` seen in #1023 for an easy traitlet alias specification.

Should we change something about this?

watsonjj, 2019-03-25:
One option is to allow the traitlet of a subclass to be found for the help message.

The downside of this option is the confusion if the subclasses have different traitlets with the same name...however this confusion already sort-of exists due to the allowed approaches shown above.

watsonjj, 2019-03-25:
Or we leave it as it is. The setting of the traitlets is some-what flexible (one can specify the base class in order to configure the subclass traitlets which share a name), but the specification of a traitlet as an alias is restrictive, and will throw an error in the help message.

kosack, 2019-03-26:
I guess the question is do we want to support using the aliases for this complex configuration, or should command-line parameters be kept simple, and we load anything complex from a config file?

watsonjj, 2019-03-26:
That is a good suggestion. We only use aliases for simple options (like file path, or Tool-specific arguments). One can see all the more complicated configuration options for the components used by the `Tool` via the `ctapipe-display-integration --help-all` command.

watsonjj, 2020-03-24:
Solution was to not use aliases for complex configuration parameters

closed 2020-03-24 by watsonjj