cta-observatory/ctapipe #1040: Separate monitoring, event, and subarray/mcheader data in the eventsource

watsonjj, 2019-04-05:


watsonjj, 2019-04-05:
Current hierarchy: 
- event.mc
- event.subarray
- event.mcheader
- event.mon
- event....

Suggestion: Have a higher level container which is passed between pipeline algorithms, which contain different sub containers which are updated at different rates:
- data.event (changes per event)
- data.mon (changes every second or so)
- data.header (changes every run/observation)
- data.subarray (or include in data.header)
- data....

kosack, 2019-04-08:
I think we should make this a more general issue, since it affects more than the data model.  I had started another one a while ago, but I think parts were also scattered in various PRs..:

See for example: #747

kosack, 2019-04-08:
strangely I can't find the original discussion in the issue tracker... It was a general discussion on data access in ctapipe, and I had suggested exactly this splitting, but gave a few other options

kosack, 2019-04-08:
I'm closing this one since it's replaced with #1041

closed 2019-04-08 by kosack