cta-observatory/ctapipe #1043: "PlanarRepresentation already defined" when using pyfact

noahhdf, 2019-04-09:
When using
```python
from fact.analysis.source import calc_off_position
```
and
```python
from ctapipe.coordinates import CameraFrame
```
astropy raises `ValueError: Representation class planar already defined`, because both packages internally define `representation.PlanarRepresentation`, which is exactly the same in both packages (in [pyfact](https://github.com/fact-project/pyfact/blob/master/fact/coordinates/representation.py) copied from [ctapipe](https://github.com/cta-observatory/ctapipe/blob/master/ctapipe/coordinates/representation.py)).

One can bypass this problem as shown [here](https://github.com/astropy/astropy/issues/8509#issuecomment-480487480). Alternatively one can push the `PlanarRepresentation` to astropy.

Further discussion in this [Astropy Issue](https://github.com/astropy/astropy/issues/8509).

noahhdf, 2019-04-09:
See corresponding [pyfact issue](https://github.com/fact-project/pyfact/issues/121)

kosack, 2019-04-17:
I think this can be closed as there is a workaround in the astropy issue here https://github.com/astropy/astropy/issues/8509#issuecomment-480487480

closed 2019-04-17 by kosack

MaxNoe, 2019-04-17:
I also renamed the corresponding class in pyfact, so no further name collisions should occure for people using both packages