cta-observatory/ctapipe #1044: Waveform sample rate

watsonjj, 2019-04-09:
Currently the only place to obtain the sample rate is from inside the mc container (`event.mc.time_slice`). This information is required to correctly calculate the pulse time. 

Where should this parameter (and other similar waveform properties such as a reference pulse shape) be stored?

kosack, 2019-04-10:
I guess in principle, it should be inside the instrument description, so in `CameraGeometry`.  It's a part of how you define the camera.  It could go in a more technical camera class (that we will add later, with module info, etc), but I think it's so fundamental to how you treat a camera image sequence that it should go in `CameraGeometry`.

kosack, 2019-04-10:
I'd suggest adding an attribute `CameraGeometry.sampling_rate` that defines it in MHz, and that field should also end up in the table metadata when you use `CameraGeometry.to_table()`

kosack, 2019-04-10:
in addition (or instead of that), the waveforms in R0-DL0 could have units attached (divide by sampling_rate, and convert to ns), but one still needs to store the sampling rate somewhere for that to work.

watsonjj, 2019-04-10:
> in addition (or instead of that), the waveforms in R0-DL0 could have units attached (divide by sampling_rate, and convert to ns), but one still needs to store the sampling rate somewhere for that to work.

Wasn't this decided against in the core developers meeting? (#684)

I agree with the decision, due to the overhead it adds to the waveform processing. We should just correctly consider the sampling rate in the charge and time extraction.

kosack, 2019-04-12:
Ok, agreed, no units attached to the waveforms, but the sampling_rate in camera should still have units.

watsonjj, 2020-03-11:
Resolved by #1142 and #1227

closed 2020-03-25 by watsonjj