cta-observatory/ctapipe #1053: CameraGeometry now recomputes the neighbor matrix for each telescope, not type

kosack, 2019-04-24:
I just ran a profile on a standard Prod3B gamma file (~2000 events, with all telescopes allowed, I think something like 400). See figure below. 

## current (slow) behavior:

Right now we are dominated by the construction of the various representations of the neighbor matrices (again, this is for a super-array of many telescopes, but it's still an issue).  The event processing is a small fraction.  I suspect this is due to some recent changes to `CameraGeometry` (when the `guess()` function was removed).  It seems now, we re-compute the neighbors *for each telescope*, which is not necessary

## old behavior

In old versions, the neighbor matrix was only recomputed for each camera *type* once, not for each camera. This was due to the `__hash__()` function returning the same hash for the same camera type, so no re-computation was made when constructing a new `CameraGeometry` that was the same as an old one, the matrix was simply re-used (memoized).  I.e. when you called `guess()` with the same inputs, you got back *the same CameraGeometry*, not a new one.

We need to revisit this behavior.

[image]

MaxNoe, 2019-04-24:
I think it's because we changed to creating instances of `CameraGeometry` in the event sources. We could add a cache there.

kosack, 2019-04-24:
Yes, I was just thinking the same, that's an easier place to do the caching.

kosack, 2019-04-24:
It was originally done in the `guess()` which was called in the EventSource, so that would be basically the same idea.

kosack, 2019-04-24:
I think it's the same for the OpticsDescription. I guess you can just put the construction of both into functions, and decorate them with `functools.lru_cache`

MaxNoe, 2019-04-24:
See #1054

MaxNoe, 2019-04-24:
Where is `neighbor_matrix_where` used? This is a python list of python lists of indices. That will be very inefficient anywhere.

MaxNoe, 2019-04-24:
It's used only in the CameraCalibrator.

MaxNoe, 2019-04-24:
And then in the `NeighborPeakWindowSum`. If this could use the matrix and not the list of indices, we would not need this.

kosack, 2019-04-24:
@watsonjj  is it possible to remove the neighbor_matrix_where and use the (sparse) matrix itself?

watsonjj, 2019-04-25:
> Where is `neighbor_matrix_where` used? This is a python list of python lists of indices. That will be very inefficient anywhere.

You have mixed up neighbor expressions. `neighbor_matrix_where` is a numpy array of shape (n_neighbor_combinations, 2). It is `neighbors` that is a "python list of python lists of indices".

The expression of the neighbors in `neighbor_matrix_where` is the most efficient expression for a loop-based approach, such as that used in `ctapipe.image.extractor.neighbor_average_waveform`, as far as I can tell. This is because the loop iterations required is kept to the minimum. Addressing other expressions of the neighbors:
1. `neighbors` is a list of lists, which as Max pointed out, is inefficient anywhere
2. `neighbor_matrix` is a numpy mask array of shape (2048, 2048) - much more entries to loop over
3. `neighbor_matrix_sparse` is a scipy sparse array, which is not compatible with numba

`neighbor_matrix_where` can be generated from `neighbor_matrix_sparse` with the following:
```python
np.column_stack([
   camera.neighbor_matrix_sparse.tocoo().row,
   camera.neighbor_matrix_sparse.tocoo().col,
])
```

But I am not sure if that is any improvement over the current calculation.

MaxNoe, 2019-04-25:
> But I am not sure if that is any improvement over the current calculation.

For the small test I made, it's over 50 times faster

MaxNoe, 2019-04-25:
Solved by merging #1054

closed 2019-04-25 by MaxNoe