cta-observatory/ctapipe #1058: TelescopeDescription string representation changed

kosack, 2019-04-25:
Old behavior:
```py3
>>> str(sub.tel[410])
'SST-1m:Digicam'
>>> str(sub[tel][100])
'SST-ASTRI:AstriCam'
```

Current Behavior:
```py3
>>> str(sub.tel[410])
'1m:DigiCam'
>>> str(sub[tel][100])
'ASTRI:AstriCam'
```

Seems the telescope type is no longer prepended. Since the full telescope identifier is used for output table naming, this causes some problems since a variable name that starts with a number (like `1m`) is forbidden.  Could we go back to the old behavior? Any objections?

kosack, 2019-04-25:
Another thing to discuss is to have these output "allowable" variable/column names, e.g:
`SST_1m_DigiCam`. The main issue there is that for optics types with no subtype (e.g. LST), is it better to get `LST_LSTCam` or `LST__LSTCam`? (in case somebody wants to split on `_` rather than lookup the details in the instrument description)

MaxNoe, 2019-04-25:
or `LST_LST_LSTCam`? :D

Also, why is it `LSTCam`, isn't it called `DragonCam`?

kosack, 2019-04-26:
>or LST_LST_LSTCam?

The name of the sub-type isn't really "LST" though, right? I guess in the real future, we'll have to think about versions of the structure, if after the prototypes, it changes. I guess it could be `<tel class>_<optics_type>_<camera>: LST_Prototype_LSTCam` (same for `MST_Prototype_NectarCam`, `SST_1m_Digicam`, `SST_ASTRI_CHEC`, etc.)

>Also, why is it LSTCam, isn't it called DragonCam?
Yes, could be also a good time to rename it - I think at the time the code was originally written, I didn't know the name, so called it LSTCam.

kosack, 2019-04-26:
There's also the fact that we have CHEC-M and CHEC-S cameras, so perhaps also having a camera subtype could be useful

GernotMaier, 2019-04-26:
Plus for prod4-SST we simulated GCT-S with 6.2 mm and with 7 mm pixels. So this should be as flexible as possible.

kosack, 2019-04-29:
> Plus for prod4-SST we simulated GCT-S with 6.2 mm and with 7 mm pixels. So this should be as flexible as possible.

Yes, we need to have version numbers inside the optics description.  Since those don't change within a prod, it's ok to not have that info in the string representation (which is so far just used to name output tables for DL1 and to name Machine Learning models).  Having the "prod" identifier somewhere more globally (like in the SubarrayDescription or the mcheader) would also be very useful.

MaxNoe, 2019-04-29:
Maybe we should aks conrad to wwrite the production info into the `target` of the run header?

RIght now, this is `Monte Carlo Beach`, which is not that helpfull :rofl:

bregeon, 2019-04-30:
> Yes, we need to have version numbers inside the optics description. Since those don't change within a prod, it's ok to not have that info in the string representation (which is so far just used to [...]

I agree on that, we do not need to be that flexible on these column names, the corresponding exact hardware configuration will be stored elsewhere, we just need minimal information here for convenience.

commit 2efcfa0a (2019-05-20): Fix broken telescope id string, Fixes #1058 (#1069)

* Fixed indexing problems with Instrument Description classes

- get_tel_ids_for_type() was broken due to change in string representation, it is now fixed.
- added many unit tests to prevent regression

* update telescope naming in muon code

* update example

* updated tutorials

* fix one more telescope string

* fixed example and tutorial

* style cleanups

* use UNKNOWN_TELESCOPE on failed guess

* update test

* added some docstrings and reformatted

* pep8 style cleanups

* renamed type to tel_type to avoid name conflict with builtin

 [... cut]
closed 2019-05-20 by kosack