cta-observatory/ctapipe #1064: EventSeeker str vs int

MaxNoe, 2019-04-30:
I think the API of the EventSeeker is not ideal. That integers mean lookup by index and str mean lookup by event id.

Pandas uses `loc` and `iloc` accessors di differentiate the too.

We could use `.idx` and `.eventid` accessors.

Opinions?

watsonjj, 2019-04-30:
I would support that. It's more explicit.

dneise, 2019-04-30:
I mentioned this already at the meeting in dortmund... and I'd like to mention it again, although it is embarrassing:

So I had a problem understanding the difference between the terms "index" and "id". I must admit up to then I lived in a world where "id" was simply an abbreviation for the word "index", but I learned this is not the case: "id" is an abbreviation for "(unique) identifier". 

Anyway I still have a hard time understanding which is which. I might be the only person having this problem ... or I might not, so I looked up the definitions of the words again ... in wikipedia ... not sure this is the right place:
 * https://en.wikipedia.org/wiki/ID
 * https://en.wikipedia.org/wiki/Index

So ... ["ID"](https://en.wikipedia.org/wiki/ID) is `Identifier, a symbol which uniquely identifies an object or record` ... can be any symbol .. 
while ["Index"](https://en.wikipedia.org/wiki/Index) (scroll down to "computer science")  is either:
1.) `Index, a key in an associative array` or
2.) `Array index, an integer pointer into an array data structure`

So ... I believe one of the definitions for "index" from wikipedia ("key in an associative array") is pretty close to the definition of "ID"... so maybe it is not only me having this problem. 

Therefore I wanted to ask again ... could we find names which are more clear here? Or at least write down our definitions of "id" and "index" somewhere in (several) obvious locations?

dneise, 2019-04-30:
So .. at the moment the definition of `id` vs `index` can e.g. be found here. 
https://github.com/cta-observatory/ctapipe/blob/master/ctapipe/io/eventseeker.py#L46

But I assume this concept of having both `id` and `index` is more fundamental than the `EventSeeker` in fact it is a cta wide concept, no?

MaxNoe, 2019-04-30:
> But I assume this concept of having both id and index is more fundamental than the EventSeeker in fact it is a cta wide concept, no?

It's more. It's a general problem whenever you have some kind of ordered data with an id, or in database speak, a `PrimaryKey`.

You can have a collection of such things, and you immediately have the problem, that you sometimes want to access things by the position in the collection and sometimes by the `Primary Key`, e.g. event_id or something else.

dneise, 2019-04-30:
Okay, so maybe we could just borrow the exact same terminology and call the accessors: `loc` and `iloc` instead of `idx` and `id`.

MaxNoe, 2019-04-30:
Yes we could, but we could make it more explicit by giving the accessors names that speak more.

dneise, 2019-04-30:
Ah so even `.by_position` and `.by_label` ?

MaxNoe, 2019-04-30:
I would call it `event_id` as this is the variable we use to make the lookup. The data container has members `obs_id` and `event_id`.

kosack, 2019-04-30:
I support this proposal.  I suppose I prefer: `seeker.event_id[12345]` and `seeker.event_index[12]` since they are the most clear.

kosack, 2019-04-30:
The only possibly reason we might want to have an array-like interface would be to allow complex indexing and slicing, like:
```py3
events = seeker.event_id[[1,15,17,23]]
events = seeker.event_index[30:40]
```
If we don't support that, then we should just stick with methods.

PR #1401 (2020-08-05): Update EventSeeker
- Replace `__get_item__` with `get_event_index` and `get_event_id` (resolves #1064)
- Always allow forward seeking (resolves #1076 & #1063)
closed 2020-08-05 by kosack