cta-observatory/ctapipe #1073: Remote test data vs ctapipe-extra

kosack, 2019-05-16:
Soon, I would like to start to keep only configuration files and defaults in ctapipe-extra (things like camera geometries, subarray layouts, etc), and move all large test files to a separate server with more space available (to avoid breaking the package size limits)

We now have a place for remote testing data.  @bregeon has uploaded the latest ctapipe-extra info to the publicly available data server, so we can start to migrate any code like notebooks that need larger test data to use that location.  This will be versioned as follows:

```
[server-base-path]/
    - ctapipe-test-data/
         - v1.0.0/
            [files and subdirs]
```

In version 1.0, this is just a copy of ctapipe-extra.  In future versions, we might organize it better.  The next step will be to make some nice utility functions in ctapipe that access this without the user having to give the full path.   In subsequent versions, we can simply symlink files that don't change to save space, and/or add new ones. 

Downloading the data works fine with astropy's `download_file` function as shown here:

[image]

kosack, 2019-05-16:
To avoid this from blowing up into an unmanageable set of files with no idea what is what, please give some suggestions for further organization of this data. E.g.:

* what test data files do we need? e.g Small sample of gammas, protoons, electrons? from which prods?, including what info (PE images?, calibration events?, etc)
* how should we name them to be clear what they are? (e.g. these could be from different productions, different arrays, etc - we need a nice way to clearly state what is used for what purpose)

kosack, 2019-05-16:
This relates to issues:
* #838 How to add LST test resources
* #738 selectable datasets
* #558 clean up tests that load data

MaxNoe, 2019-05-16:
Looks good! Can we make a small wrapper function that uses our server by default and a different storage location?

kosack, 2019-05-16:
> Looks good! Can we make a small wrapper function that uses our server by default and a different storage location?

Yes, I think the best way to start is something similar to the `get_dataset_path` function in ctapipe, something like `fetch_test_dataset(name)`. 

It reminds me of another issue I would like to bring up: having some notion of global "Services", which are basically `Components` who get instantiated and configured every time you load the ctapipe module (where the configuration comes from some default `.ctapiperc` file or similar, or from a Tool's configuration system if it exists). These would be things like this test data fetcher (configurable version and server path), the provenance system (with some configuration options), and for example a random number generator (with configurable seed).

So in that case you would have something like the following in .ctapiperc, or your config file, or command-line config (where the defaults would be used if not specified):
```yaml
DataSets:
    server_url: https://xxx.xxx.xxx.xxx/
    version: v1.0.0
```
and in code you would do something like:
```py3
from ctapipe.core.services import dataset
# dataset is a pre-instantiated and configured DataSet Component
filename = dataset.fetch("gamma_test_large.simtel.gz")

dataset.version = 'v1.0.1'
filename2 = dataset.fetch("gamma_test_large.simtel.gz") # might be different, due to version change

```

kosack, 2019-05-16:
I think we should also think about a sub-directory structure for each version, rather than a flat list of many files. That could make the organization better.

bregeon, 2019-05-17:
Some comments (in reverse order):
- sub-directories: for the main use case, unit tests, I would suggest to create subdirectories following the modules that are tested, i.e. each module has its own sub-directory with its tests files.
- **dataset service**: that is a great (very much Gaudi like :)) idea, that should be addressed in a separate issue though, and I would suggest to have a such a service for calibrations files
- **describing the content**: I propose that each file be described with a "Provenance" like [dictionary](https://github.com/cta-observatory/ctapipe/blob/master/ctapipe/core/provenance.py), with information such as owner, version, list of target unit tests, file size... we could have one big file with all the dictionnaries (a list of dict). And I also suggest we have a release notes like file with just a line describing what's in/changed in each version.
- **new test files**: I would suggest to gather all the test files proposed in the different issues and PRs reference above (#558, #738, #838, Hipehdf5, muon test files... I also have some Lidar test files) and to add these immediately on the server so that thes things can go forward.
- **backup**: I'll backup each version of the files on sps at CC, but in the final setup we'll need a mirror.

vuillaut, 2019-05-17:
Hi
- I am not sure that files should follow the modules organisation. I can imagine many files being used for very different unit tests (as it's already the case for the `gamma_test` file.
- Related comment for provenance. Unit-tests are targets, not really provenance? Files are provided and then unit-tests are developed. Ensuring to go backward to update the list of unit-tests for each file will be tiresome.
- Would we eventually use the same files for cta-benchmarks?

bregeon, 2019-05-17:
Other sub-directory structure could be:
- by instrument: Full array North, Full array South, LST, MST, SST, Lidar, Hess, Magic, etc...
- by "Prod": Prod2, Prod3, Prod3b, Prod4, LST+Magic, etc...
- by "particle": gamma, electron, proton, calibration, etc...

And yes, I think we would mostly use the same files for benchmarks (we could have a special benchmark sub-directory :)).

kosack, 2019-05-17:
> describing the content: I propose that each file be described with a "Provenance" like dictionary, with information such as owner, version, list of target unit tests, file size... we could have one [...]

This is actually something I've been defining for CTAO.  See Slide 18/19 of https://indico.cta-observatory.org/event/2114/contributions/19323/attachments/15129/19143/dpps_aswg.pdf 

I recommend we start to attach these metadata to all files we produce, or provide. 

An example of these headers in an ECSV file (they can also be stored as a YAML or JSON hierarchy rather than a flat list)
```
# %ECSV 0.9
# ---
# datatype:
# - {name: A, unit: m, datatype: float64}
# - {name: B, unit: s, datatype: float64}
# meta: !!omap
# - {CTA METADATA VERSION: 2}
# - {CTA CONTACT ORGANIZATION: CTAO}
# - {CTA CONTACT NAME: CTAO Support}
# - {CTA CONTACT EMAIL: support@cta-observatory.org}
# - {CTA PRODUCT DESCRIPTION: Cat C DL3 event list}
# - {CTA PRODUCT CREATION_TIME: '2018-11-10 15:30:00'}
# - {CTA PRODUCT ID: TBD}
# - {CTA PRODUCT DATA CATEGORY: C}
# - {CTA PRODUCT DATA LEVEL: DL3}
# - {CTA PRODUCT DATA ASSOCIATION: SUB}
# - {CTA PRODUCT DATA TYPE: EVT}
# - {CTA PRODUCT DATA SUBTYPE: ''}
# - {CTA PRODUCT DATA MODEL NAME: open-gamma-astro}
# - {CTA PRODUCT DATA MODEL VERSION: v0.2}
# - {CTA PRODUCT DATA MODEL URL: 'https://github.com/open-gamma-ray-astro/gamma-astro-data-formats/tr
# - {CTA PRODUCT FORMAT: fits}
# - {CTA PROCESS TYPE: observation}
# - {CTA PROCESS SUBTYPE: standard}
# - {CTA PROCESS ID: '12345'}
# - {CTA ACTIVITY NAME: pipeline-stage-3}
# - {CTA ACTIVITY TYPE: software}
# - {CTA ACTIVITY ID: 5367fcf8-e75f-11e8-9692-3c15c2d6877e}
# - {CTA ACTIVITY START: '2018-11-10 15:24:11'}
# - {CTA ACTIVITY END: '2018-11-10 15:29:00'}
```

maxnoe, 2021-02-19:
We now switched to downloading files from the data server

closed 2021-02-19 by maxnoe