cta-observatory/ctapipe #1076: EventSeeker no longer works with SimTelArray files

kosack, 2019-05-22:
```py3
source = event_source("events.simtel.gz")
seeker = EventSeeker(source)                                                                    
```
throws an exception:
```
OSError: Reader is not compatible as input to the event_source is a stream
   (seeking not possible)
```

But I thought the whole point of the EventSeeker class was to provide a way to seek forward (or backward) in a file to a particular event, by event count or event_id?  Even if the file is not backward-seekable (a stream for example), forward seeking should still work, and backward seeking can be implemented by re-opening the stream (or throwing an exception as now). 

In other words I think that forward-seeking should work fine for a SimTelArray file, but backward seeking should either fail or work in a non-optimal way. Is there a reason this changed?  It's quite useful for event-wise debugging, so having this functionality back is important. 

A use case:
- user identifies a reconstruction problem in the DL1 or DL2 data, and identifies a set of event_ids that exhibit the problem
- to debug, the user wants to open a raw data file, seek to each of the identified events, and plot the raw camera data. 

Of course one could write a loop manually to do this each time, but EventSeeker was supposed to be a common way.

watsonjj, 2019-05-22:
It seems a change was made to `SimtelEventSource` in which passing `back_seekable=True` is required to go to earlier events in the file

watsonjj, 2019-05-22:
But yes, some changes could be made to `EventSeeker` to allow streams + seeking forward

kosack, 2019-05-24:
Is the user allowed  to set `back_seekable`?  I assumed that is not under the user's control, and should be set correctly by the EventSource depending on the file type (e.g. if it's a stream vs a tabular file)?

MaxNoe, 2020-03-19:
@kosack in case of the `SimtelEventSource` together with gzip files, the user can request to use the python gzip module instead of zcat which is a bit slower but allows back seeking

PR #1401 (2020-08-05): Update EventSeeker
- Replace `__get_item__` with `get_event_index` and `get_event_id` (resolves #1064)
- Always allow forward seeking (resolves #1076 & #1063)
closed 2020-08-05 by kosack