cta-observatory/ctapipe #1077: provide some numba  versions of standard algorithms

kosack, 2019-05-22:
This is not required at this point, but still might be interesting to test

Some functions like `tailcuts_clean` (and other cleaning variants), and `hillas_parameters`  are called often (once per event per telescope) and could probably be sped up without major changes to the code, and even without using an external hand-coded library like *hipeCTA*, using only Numba.  This worked well with the charge extraction code, and could be extended to other image functions. 

This may also be useful to the teams doing realtime-analysis prototyping. 

## Example:
Right now, simply applying numba to e.g. `tailcuts_clean` has no effect, since the inputs are a python object (CameraGeometry) and a numpy array, e.g. the function signature is 
```py3
def tailcuts_clean(geom : CameraGeometry, image: np.ndarray) -> np.ndarray:
    ...
```

Since numba doesn't know what to do with `CameraGeometry`, it can't optimize. Splitting this into two functions with simple numpy inputs to one would allow it to work:
```py3
def tailcuts_clean(geom, image):
    return _tailcuts_clean_fast(
        neighbor_matrix=geom.neighbor_matrix_sparse, 
        image=image
    ) 

@numba.njit(parallel=True)
def _tailcuts_clean_fast(...):  # this takes only numpy arrays and is optimizable
    # basically a cut-paste of the current code, after "unpacking" the CameraGeometry
```
This would also allow us to broadcast this same optimized function to a large set of images if in the future we load more than one event at a time. 

## Where applicable:
This pattern could be applied at least to the following functions (probably more), but testing on a few would be interesting.
* `ctapipe.image.cleaning`:
  * `tailcuts_clean()`  (good place to start with a test) 
  * `dilate()`
  * `number_of_islands()`
  * `apply_time_delta_cleaning()`
  * `fact_image_cleaning()`
* `ctapipe.image.hillas.hillas_parameters()`
* `ctapipe.image.timing.timing_parameters()`
* `ctapipe.image.leakage.leakage_parameters()`

kosack, 2019-05-22:
for reference, Numba currently supports optimization of many numpy functions shown here: https://numba.pydata.org/numba-doc/dev/reference/numpysupported.html#functions

kosack, 2019-05-22:
one problem that may block this from being so simple is that numba doesn't yet natively support `scipy.sparse.csr_matrix`, which we do use for the neighbor storage. So either it has to be expanded into a non-sparse matrix first (`m.todense()`) , or the for-loops done explicitly as in https://stackoverflow.com/questions/19420171/sparse-matrix-in-numba

maxnoe, 2020-10-26:
I reimplemted many of our algorithms using numba, where it delivered a good speed improvement. See for example #1361.

I'll close this for now, as there are no low-hanging fruits left I think.

closed 2020-10-26 by maxnoe