cta-observatory/ctapipe #1080: Configuration hierarchies

MaxNoe, 2019-05-27:
We should think about and then define which configuration options we want to have and in which hierarchies, to be then able to implement it.

For example, from least specific to most specific:

* Global value
* Value per subarray
* Value per telescope / camera *type*
* Value per telescope / camera *id*

kosack, 2019-05-28:
we want to support all of those. The question is exactly how... It think there are some scattered discussions in other issues related to this.   Probably the simplest is to allow a default value to be defined as well as telescope id or type-wise values. How fancy we want to make that is still a question...  Here's one idea allow either a single value  or a dict of specialized values for all options, so that one can specialize any option:

```json
{
    "a_global_parameter" : 6.0,
    "some_analysis_threshold": {
         "default": 1.0,
         "MST_MST_NectarCam": 2.0,
         "MST_MST_FlashCam": 1.8,
         "TEL021" : 0.6,  # allow mixing of type and ID? require ordered dict
    }
}
```
But that requires that order is preserved (e.g. do you apply default, then look for type, then look for specific telescope?).

MaxNoe, 2019-05-28:
I would not use that flat hierarchy, by maybe:

```yaml
some_analysis_threshold:
    default: 5
    telescope_type:
        MST_MST_FlashCam: 1.5
    telescope_id:
        1: 0.5
```

kosack, 2019-05-28:
Sounds good. In a hierarchy like that the order of operations for applying an option must be always:
1. default
2. telescope_type (overrides 1)
3. telescope_id (overrides 1,2)

I think that's ok, since it goes from most general to most specific. Anything else would be confusing.

kosack, 2019-05-28:
perhaps name the sub-hierarchy `for_telescope_type` and `for_telescope_id` so that it reads as an english sentence. E.g. "threshold for_telescope_type: X = Y"

MaxNoe, 2019-05-28:
This would also add another point against json: json only allows string as keys

watsonjj, 2019-10-26:
Implemented in #1129

closed 2019-10-26 by watsonjj