cta-observatory/ctapipe #1087: Why data volume reduction before calibration?

FrancaCassol, 2019-06-13:
Hi,
I am creating a daughter of the CameraCalibrator class dedicated to the calibration of LST real data (in cta-lstchain). Looking at the implementation of CameraCalibrator I am a bit surprised to see that data volume reduction is performed in  _calibrate_dl0() before the charge calibration (in _calibrate_dl1). What is here meant for volume reduction? Because in general It seems to me that data reduction on calibrated data should be more precise. So, I would do the other way around: calibrate the dl0 and reduce the dl1 ...

watsonjj, 2019-06-13:
You are right that a form of charge extraction and camera calibration is likely needed for the data volume reduction - to identify pixels which have signal. However, as I understand it, the data volume reduction will be done before ctapipe - the DL0 waveforms ctapipe will read will already have data volume reduction applied. 

The data volume reduction is done online, and therefore may utilise a rough, fast charge extraction and camera calibration that is not perfect. Whereas for extracting the images - we require accurate charge extraction and calibration.

And its important to remember that the output of the data volume reduction step is waveforms (at least for the pixels containing signal). Ctapipe will then need to read these waveforms, and perform the "best" charge extraction and calibration (which may be slower than the methods used in the data volume reduction).

watsonjj, 2019-06-13:
Data volume reduction in this context is to do with reducing the amount of waveform data stored to disk.

There is a cleaning step after charge extraction (tailcuts etc.) which one may consider is "reducing the data volume of images". But this is a different operation to what is referred to as data volume reduction.

FrancaCassol, 2019-06-13:
Hi @watsonjj,

yes, I agree that most probably the gain reduction will be done at the event builder level, but, as you say, any type of reduction will  ask for a calibration (even if rough), so if we decide to introduce a waveform reduction in ctapipe at this level why should we put it before the calibration? Is there a specific reason? (how do you think to do the reduction here with uncalibrated data?)

I thinks the problem is also coming from some inconsistency with the present data model in ctapipe, if you look at the DL0 container it speaks of calibrated waveforms:

```
class DL0CameraContainer(Container):
    """
    Storage of data volume reduced dl0 data from a single telescope
    """
    trigger_time = Field(None, "Telescope trigger time, start of waveform "
                               "readout")
    trigger_type = Field(0o0, "camera trigger type")

    waveform = Field(None, (
        "numpy array containing data volume reduced "
        "p.e. samples"
        "(n_pixels, n_samples). Note this may be a masked array, "
        "if pixels or time slices are zero-suppressed"
    ))
```

This is different from what @kosack suggests in his data model document where the DL0 level contains both the waveform and the image in digital counts: 

[image]

If we follow this proposal both the image and waveform reduction (if not yet done at event builder level) must be at DL1 level with calibrated data, which in turn is consistent with the present DL1 data model in ctapipe which foresees the gain selection at this level

```
class DL1CameraContainer(Container):
    """
    Storage of output of camera calibration e.g the final calibrated
    image in intensity units and the pulse time.
    """
    image = Field(
        None,
        "Numpy array of camera image, after waveform extraction."
        "Shape: (n_chan, n_pixel)"
    )
    pulse_time = Field(
        None,
        "Numpy array containing position of the pulse as determined by "
        "the extractor."
        "Shape: (n_chan, n_pixel, n_samples)"
    )
    #TODO: Remove when gain selection added?
    gain_channel = Field(
        None,
        "boolean numpy array of which gain channel was used for each pixel "
        "in the image "
    )
```

@karl, since we are in process to setup the DL0-DL1 production in cta-lstchain,  could we update the DL0 (DL1?) ctapipe container so as to use the most recent/valuable solution for the LST1 data?

watsonjj, 2019-06-13:
Ctapipe will read DL0 waveforms. These are the waveforms saved to disk, and due to their large data volume, they need to be reduced smartly.

Currently as a prototype, ctapipe also reads R1 waveforms. **R1 waveforms have the camera-specific calibration applied to them, but no data volume reduction. They are calibrated waveforms.** The data volume reduction step in ctapipe can be used to test algorithms for the data volume reduction step. This transforms the R1 waveforms into DL0 waveforms.

DL0 waveforms are then processed by the Image extractors to get the single values per pixel which make up the DL1 image.

It doesn't make sense to do the data volume reduction after the DL1 step, because beyond the DL1 calibration you no longer have waveforms. The purpose of the data volume reduction is to reduce the size of waveforms so that they can be stored (and archived) on disk.

kosack, 2019-06-17:
> DL0 level contains both the waveform and the image in digital counts

Those are not really digital counts, but calibrated values (pedestals and gains applied) stored as integers (so losing some precision, but enough to do data volume reduction and basic Hillas parameterization) .  Right now we do not model this in ctapipe. Probably we should at some point, but I was waiting until the on-line camera calibration procedures are defined.

FrancaCassol, 2019-06-18:
@kosack,
perfect, this is what I thought it should be (perhaps the unit DC is a bit misleading ). So, the charge calibration mentioned at the end of _calibrate_dl1() is a second step calibration?

```
def _calibrate_dl1(self, event, telid):
        waveforms = event.dl0.tel[telid].waveform
        if self._check_dl0_empty(waveforms):
            return
        n_samples = waveforms.shape[2]
        if n_samples == 1:
            # To handle ASTRI and dst
            # TODO: Improved handling of ASTRI and dst
            #   - dst with custom EventSource?
            #   - Read into dl1 container directly?
            #   - Don't do anything if dl1 container already filled
            #   - Update on SST review decision
            corrected_charge = waveforms[..., 0]
            pulse_time = np.zeros(waveforms.shape[0:2])
        else:
            # TODO: pass camera to ImageExtractor.__init__
            if self.image_extractor.requires_neighbors():
                camera = event.inst.subarray.tel[telid].camera
                self.image_extractor.neighbors = camera.neighbor_matrix_where
            charge, pulse_time = self.image_extractor(waveforms)

            # Apply integration correction
            # TODO: Remove integration correction
            correction = self._get_correction(event, telid)[:, None]
            corrected_charge = charge * correction

        event.dl1.tel[telid].image = corrected_charge
        event.dl1.tel[telid].pulse_time = pulse_time

        # TODO: Add charge calibration

```

kosack, 2019-06-18:
I think you're not wrong, but perhaps we are also talking about different things: analyzing simulations in a similar way to real data means a few extra steps. The real sequence would be something like:

1. In the camera server:
     1. apply basic calibration (to integer level) and output R1
1. In the ACADA data handling system:
     1. read R1 and apply data volume reduction and store as DL0
     1. calculate rough hillas parameters and send to realtime pipeline
1. in the offline pipeline (Cat-B or Cat-C):
     1. read DL0 and either 
        1.  un-apply basic calibration and apply better calibration
        1. apply corrections to basic calibration
    1. etc.

We need to simulate the first 2 steps for Monte-Carlo data.

FrancaCassol, 2019-06-18:
Yes, the point is that we don't have ACADA now (but perhaps I miss something, is there an ACADA repository/code somewhere?), so with the prototype data we are using ctapipe also for doing steps 1 and 2 ... this is probably at the origin of the confusion.

I totally agree that we must test these two points with MC data, for that we need to have the MON container also when we read the MC data. Waiting  for discussion in  [https://github.com/cta-observatory/ctapipe/issues/1041]() to converge, could we  modify the DataContainer() in the MC readers with a EventAndMonDataContainer(), so that one can store the results of the calibration procedure also when performed with MC events?

vuillaut, 2019-06-19:
> I think that sequence looks fine, but perhaps we are also talking about different things: analyzing simulations in a similar way to real data means a few extra steps. The real sequence would be something like:
> 
> 1. In the camera server:
>    
>    1. apply basic calibration (to integer level) and output R1
> 2. In the ACADA data handling system:
>    
>    1. read R1 and apply data volume reduction and store as DL0
>    2. calculate rough hillas parameters and send to realtime pipeline
> 3. in the offline pipeline (Cat-B or Cat-C):
>    
>    1. read DL0 and either
>       
>       1. un-apply basic calibration and apply better calibration
>       2. apply corrections to basic calibration
>    2. etc.

Hi @kosack 

Am I understanding correctly?
- R1 = integers
- DL0 = integers

--> un-apply calibration on DL0 is not an exact reverse operation (some precision loss as data is stored in integers)

kosack, 2019-06-19:
Yes both R1 and DL0 should store waveforms as 16-bit unsigned integers (with some metadata fields to tell the scale and shift to apply to make them floating point). E.g. `waveform_value = WAVE_SCALE * (waveform_int - WAVE_SHIFT)` . And that value is then in units of PE  or equivalent (at least roughly calibrated).

kosack, 2019-06-19:
This is somewhat confusing since we have not yet finalized the R1 and DL0 interface definitions in CTAO - this is right now just a rough guess of what the final version will be, but it's reasonable I think, and useful to try out to see if there are problems.  Once there is an official interface control document, we will have to implement exactly the data model associated with it .

maxnoe, 2020-10-26:
I think is discussion is settled with the data format document.

closed 2020-10-26 by maxnoe